TFyre/bambu-farm #3: Printer notifications obscure the interface

wieckowy, 2024-01-24:
Notifications sent by the printer cover the entire interface.
[image]

TFyre, 2024-01-24:
If you click on them, they go away :)

Is that ok?

wieckowy, 2024-01-24:
Yes, of course :) But these notifications could be much smaller and appear in the corner of the screen.

TFyre, 2024-01-24:
This will go out with the next update, let me know if its better/worse :)

[image]

wieckowy, 2024-01-24:
Amazing, thanks!
[image]

PS - I confirm that uploading files and printing works in Bambu A1.

closed 2024-01-24 by TFyre