TFyre/bambu-farm #7: "Set target temperature" not work

wieckowy, 2024-01-25:
Hi,
I'm testet "Set target" on A1 and P1S for hotend and headbed - unfortunately this option does not work for A1 and P1S.

I also tested sending custom gcode in A1 - this option works properly.

PS - unfortunately this option does not work for A1 and P1S.

I also tested sending custom gcode in A1 - this option works properly.

A good idea is to add small buttons that allow you to quickly warm up the printer to one of the predefined values (this is more convenient than expanding the menu with the right mouse button):

[image]

 It would be nice if the user could define these values ​​himself in the settings.

wieckowy, 2024-01-25:
Update
[image]

TFyre, 2024-01-26:
Found the bug, will be solved in the next release...

Leaving the issue open for UI related items

TFyre, 2024-02-07:
I went a bit in a different direction with this one. For more details check #17 

Since mouse clicks are now normal left clicks, is this ok? Can we close the ticket or do we still need the quick buttons?

TFyre, 2024-02-28:
Please re-open the ticket if required

closed 2024-02-28 by TFyre