TFyre/bambu-farm #26: X1C with more than 1 AMS not showing filament running

blasirio, 2024-02-17:
In the pic, it was running black filament.

[image]

[image]

TFyre, 2024-02-17:
1) looks like your video is not working, have you checked out the details on the docker solution for video on X1C: https://github.com/TFyre/bambu-farm/tree/main/docker/bambu-liveview

2) can you please go to "Show  Logs" for this printer, in the filter box put "tray_now", select the first message and share the text from the "RAW" box. You can replace any identifiable information like serial numbers / ip addresses etc with "REDACTED"

TFyre, 2024-02-17:
Please try version 1.0.8 and let me know 😄

commit 022f47e3 (2024-02-17): BUG: fix active filament for multi AMS (fixes #26)
closed 2024-02-17 by TFyre