TFyre/bambu-farm #30: printers do not show idle

Jostxxl, 2024-02-28:
Hello there,
thanks so far for this awesome interface.
We are in the middle of setting it up but ran into a problem with batch printing.
Most printers do not show idle in the list although they are idling.
They show local or cloud even though they have no active print.
We tried to force the stop and request full status but it didnt work.

Any ideas what this might cause?

This is the only related error i could see:
`2024-02-28 01:24:15,284 ERROR [com.tfy.bam.pri.BambuPrinterStream] (vert.x-eventloop-thread-2) 899 Killian: clientFailure: io.netty.channel.AbstractChannel$AnnotatedConnectException - Connection refused: getsockopt: /192.168.188.43:6000`

TFyre, 2024-02-28:
3. The RAW output has `"print_type" : ""` in it... I was expecting `idle` to be in there... I will have to find a different "trigger" to sense if the printer is done printing

TFyre, 2024-02-28:
Please test with https://github.com/TFyre/bambu-farm/releases/tag/v1.1.0

commit fb725551 (2024-02-28): Implement GCodeState enum (fixes #30)