TFyre/bambu-farm #31: suggestion: do not show livestream area when livestream is not activated

Jostxxl, 2024-02-28:
Hello there!
We have several printers set up in the dashboard and since were not using the liveview (yet) is there any way to hide the live feed area?

Other ways to declutter the dashboard would also be nice.

[image]

TFyre, 2024-02-28:
Thanks for reporting the issue!

Right now there is no way to disable liveview (since this was 1 of the biggest reasons for building the software 😄)

As a temporary workaround, you can add the following to `styles.css` ([README#Custom-CSS](https://github.com/TFyre/bambu-farm?tab=readme-ov-file#custom-css)) and restart the application:
```css
.dashboard-view div.image {
    display: none;
}
```

If you need this as a feature, please describe how you would like it to work? Disabling it for a single printer will make the interface look "broken" with missing "blocks"... do you want to globally disable it for all printers?

How many printers of which model do you have in the farm?

Jostxxl, 2024-02-28:
Yes, thats exactly what i meant. A feature where you can globally disable it for all printers. I currently Have 21 P1P and 39 X1C in the Farm

shotaxx, 2024-02-28:
> As a temporary workaround, you can add the following to `styles.css` ([README#Custom-CSS](https://github.com/TFyre/bambu-farm?tab=readme-ov-file#custom-css)) and restart the application:
> 
> ```css
> .dashboard-view div.image {
>     display: none;
> }
> ```

This worked flawlessly and is totally sufficient for us. This way we can modify the dashboard visuals to our needs and likings.

TFyre, 2024-02-28:
Im thinking of building it with 3 settings:

```properties
# defaults to true, when false, disables live view globally
bambu.live-view=true

# defaults to true, when false, disables live view for dashboard, but will still be available in detail view
bambu.dashboard.live-view=true

# defaults to true, when false, disables per printer
bambu.printers.myprinter1.stream.enable=true
```

Let me know if this will work for you or if it needs more explanation?

TFyre, 2024-02-28:
> This worked flawlessly and is totally sufficient for us. This way we can modify the dashboard visuals to our needs and likings.

this will hide it client side, but i suspect in the background the data still flows 😄

shotaxx, 2024-02-28:
> Im thinking of building it with 3 settings:
> 
> ```ini
> # defaults to true, when false, disables live view globally
> bambu.live-view=true
> 
> # defaults to true, when false, disables live view for dashboard, but will still be available in detail view
> bambu.dashboard.live-view=true
> 
> # defaults to true, when false, disables per printer
> bambu.printers.myprinter1.stream.enable=true
> ```
> 
> Let me know if this will work for you or if it needs more explanation?

This sounds very good! 

> > This worked flawlessly and is totally sufficient for us. This way we can modify the dashboard visuals to our needs and likings.
> 
> this will hide it client side, but i suspect in the background the data still flows 😄

Oh yeah that could still overload our system i guess :D
If you have Discord, you could send me an invite, maybe its easier to talk about our use case there.
Name: "shotaxx"

TFyre, 2024-02-28:
Cannot add you as a friend on Discord, says you not accepting friends :) Please add me instead: tfyre

TFyre, 2024-02-28:
Please test with https://github.com/TFyre/bambu-farm/releases/tag/v1.1.0

shotaxx, 2024-03-01:
That does work when you are not using docker for the livestream at all. However, when you use docker, it throws this exception:
`
2024-03-01 20:42:43 Configuration validation failed:
2024-03-01 20:42:43     SRCFG00050: bambu.remote.view in EnvConfigSource[source=file:/.env] does not map to any root
2024-03-01 20:42:43     SRCFG00050: bambu.dashboard.remote.view in EnvConfigSource[source=file:/.env] does not map to any root
2024-03-01 20:42:43 java.lang.RuntimeException: Failed to start quarkus
2024-03-01 20:42:43     at io.quarkus.runner.ApplicationImpl.doStart(Unknown Source)
2024-03-01 20:42:43     at io.quarkus.runtime.Application.start(Application.java:101)
2024-03-01 20:42:43     at io.quarkus.runtime.ApplicationLifecycleManager.run(ApplicationLifecycleManager.java:111)
2024-03-01 20:42:43     at io.quarkus.runtime.Quarkus.run(Quarkus.java:71)
2024-03-01 20:42:43     at io.quarkus.runtime.Quarkus.run(Quarkus.java:44)
2024-03-01 20:42:43     at io.quarkus.runtime.Quarkus.run(Quarkus.java:124)
2024-03-01 20:42:43     at io.quarkus.runner.GeneratedMain.main(Unknown Source)
2024-03-01 20:42:43 Caused by: io.quarkus.runtime.configuration.ConfigurationException: Failed to read configuration properties
2024-03-01 20:42:43     at io.quarkus.deployment.steps.RuntimeConfigSetup.deploy(Unknown Source)
2024-03-01 20:42:43     ... 7 more
`

I suppose it is not linked in the docker configuration part? When leaving it only in .env the application ignores that since it has to be configured in bambu-web-env.txt when using docker right?

TFyre, 2024-03-01:
```
SRCFG00050: bambu.remote.view in EnvConfigSource[source=file:/.env] does not map to any root
```

it should be `bambu.remote-view`

shotaxx, 2024-03-06:
Yeah that was my fault. Everything regarding this topic is working now! Thank you!
On behalf of Jostxxl (im working with him) this can be closed as completed.

closed 2024-03-06 by TFyre