TFyre/bambu-farm #47: LiveView - No One Is Publishing to Path

Pedalnetics, 2024-03-11:
I have 2 X1C's setup in my configs, but LiveView isn't working.  I've confirmed "LAN Mode LiveView" is enabled via the printer touchscreen on both units, and have gone back over my configs, access codes, IP's, and serial numbers, and can't find an error compared to the docs.

The UI works otherwise, I can see the printer/status/AMS contents/etc, so the IP & access code seems to be correct.

I've also validated the "gotcha" of using the PRINTER_ID in the compose file and ensured that lines up.

```
mediamtx_1  | 2024/03/11 17:28:39 INF [WebRTC] [session adb083dd] created by 172.18.0.6:56950
mediamtx_1  | 2024/03/11 17:28:39 INF [WebRTC] [session adb083dd] closed: no one is publishing to path 'x1c2'
mediamtx_1  | 2024/03/11 17:28:39 INF [WebRTC] [session 124e662e] created by 172.18.0.6:56950
mediamtx_1  | 2024/03/11 17:28:39 INF [WebRTC] [session 124e662e] closed: no one is publishing to path 'x1c1'
```

I did a `docker logs` on one of the liveview containers and see this:

```
ffmpeg version 6.1.1 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.4.0-1ubuntu1~22.04)
  configuration: --disable-debug --disable-doc --disable-ffplay --enable-alsa --enable-cuvid --enable-ffprobe --enable-gpl --enable-libaom --enable-libass --enable-libfdk_aac --enable-libfreetype --enable-libkvazaar --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libopus --enable-libplacebo --enable-librav1e --enable-libshaderc --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpl --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-nonfree --enable-nvdec --enable-nvenc --enable-cuda-llvm --enable-opencl --enable-openssl --enable-stripping --enable-vaapi --enable-vdpau --enable-version3 --enable-vulkan
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
[tls @ 0x5652b7cb1340] Unknown error
[in#0 @ 0x5652b7cae380] Error opening input: Input/output error
Error opening input file rtsps://bblp:<ACCESSCODE>@<PRINTERIP>:322/streaming/live/1.
Error opening input files: Input/output error
Exiting normally, received signal 15.
```
Any suggestion on what to try next? It looks like maybe some SSL error based on that TLS line's "Unknown error" output

Pedalnetics, 2024-03-11:
Running Ubuntu 22.04.
322 connects, but no SSL cert response; 8883 responds with an SSL cert:

```
root@bambufarm:~# openssl s_client -connect 192.168.19.115:322
CONNECTED(00000003)
<Waited 60seconds, nothing>
^C
```

```
root@bambufarm:~# openssl s_client -connect 192.168.19.115:8883
CONNECTED(00000003)
Can't use SSL_get_servername
depth=1 C = CN, O = "BBL Technologies Co., Ltd", CN = BBL CA
verify error:num=19:self-signed certificate in certificate chain
verify return:1
depth=1 C = CN, O = "BBL Technologies Co., Ltd", CN = BBL CA
verify return:1
depth=0 CN = <CN>
verify return:1
---
Certificate chain
 0 s:CN = <CN>
   i:C = CN, O = "BBL Technologies Co., Ltd", CN = BBL CA
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: Mar  2 01:19:39 2023 GMT; NotAfter: Feb 27 01:19:39 2033 GMT
 1 s:C = CN, O = "BBL Technologies Co., Ltd", CN = BBL CA
   i:C = CN, O = "BBL Technologies Co., Ltd", CN = BBL CA
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: Apr  4 03:42:11 2022 GMT; NotAfter: Apr  1 03:42:11 2032 GMT
---
Server certificate
-----BEGIN CERTIFICATE-----
<Cert hash>
-----END CERTIFICATE-----
subject=CN = <CN>
issuer=C = CN, O = "BBL Technologies Co., Ltd", CN = BBL CA
---
No client certificate CA names sent
Peer signing digest: SHA512
Peer signature type: RSA
Server Temp Key: ECDH, prime256v1, 256 bits
---
SSL handshake has read 2278 bytes and written 419 bytes
Verification error: self-signed certificate in certificate chain
---
New, TLSv1.2, Cipher is ECDHE-RSA-AES256-GCM-SHA384
Server public key is 2048 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-RSA-AES256-GCM-SHA384
    Session-ID: <SESSION>
    Session-ID-ctx:
    <key/PSK>
    Start Time: 1710182878
    Timeout   : 7200 (sec)
    Verify return code: 19 (self-signed certificate in certificate chain)
    Extended master secret: no
---
^C
root@bambufarm:~#
```

TFyre, 2024-03-11:
OK, this is related to `LAN Mode LiveView` not working as expected... Please disable/enable LiveView on the printer and/or power cycle the printer (switch on/off) and then try again, without viewing it from bambu/orca slicer...  Ive seen reports where people indicate that its not very stable doing it multiple times (connecting / disconnecting to liveview)

TFyre, 2024-03-11:
output from the openssl command for port 322 and 8883 should be identical apart from session details

TFyre, 2024-03-11:
Awesome! Let me know what you find :)

TFyre, 2024-03-11:
Would love it if you can share some screen recordings that i can add to youtube like https://youtube.com/@TFyre

Pedalnetics, 2024-03-12:
@TFyre FYI My printer liveview that was working yesterday, I've not sent any prints to it nor viewed it from anything else, and it's broke as of an hour or so ago. Nothing useful in the logs whatsoever, back to the same original error I found in docker logs yesterday. Power cycle fixed it again, so I'm guessing this is a bug/error/problem on the Bambu side.
However I do have another printer doing liveview via Home Assistant, and it's camera has been stable for over a week.  I'm not sure if maybe it's how the camera is being accessed that's the issue/if it's worth peeking at the HA code and see if they have any tricks: https://github.com/greghesp/ha-bambulab

TFyre, 2024-03-12:
Are we comparing apples with apples here? Like are both printers x1c's?

The camera stream is a known issue with x1c... ive been testing with another user where it looks like intermittent wifi connectivity veing the issue...

Generally ffmpeg streaming is really stable...

Pedalnetics, 2024-03-12:
Yup, 2 X1C's on BambuFarm, 1 X1C on HomeAssistant.  My Wifi AP is ~30feet from the printers, ceiling mounted, and wifi client map shows a strong single from all 3 X1C's with connectivity-uptimes of quite a while minus the one I rebooted yesterday (and it's uptime shows it hasn't disconnected/reconnected since that rebot).
I'll go poke around some more docker logs and see if I can find anything/catch it happening again.

Pedalnetics, 2024-03-13:
Interesting; both cameras broke with the same ffmpeg tls errors, and then both self-healed with zero intervention; no docker reboot, no printer setting toggles or reboots.

TFyre, 2024-06-03:
Ive added a `timeout` to ffmpeg to make it more robust when power cycling the printer or if it drops from the network.

Check out #87

tommy4111, 2024-06-15:
I'm getting the same issue.  Any thing I should check?  Gone through all the settings, not sure what I'm missing.

TFyre, 2025-01-04:
Closing stale issue, please re-open if required

closed 2025-01-04 by TFyre