TFyre/bambu-farm #50: Printer stopped updating, java stacktrace

Pedalnetics, 2024-03-12:
One of my printers stopped updating. In the console session the docker-compose was ran it it's showing the same Java stacktrace over and over:
`bambuweb_1  | 2024-03-11 23:47:46,155 ERROR [com.tfy.bam.pri.BambuPrinterImpl] (MQTT Call: camel-paho-35e24b0959da6fc9) Cannot build message: Expect message object but got: 6304 - {"print":{"ams":{"ams":[{"humidity":"3","id":"0","temp":"23.9","tray":[{"id":"0"},{"id":"1"},{"id":"2"},{"bed_temp":"0","bed_temp_type":"0","cali_idx":-1,"cols":["161616FF"],"ctype":0,"drying_temp":"0","drying_time":"0","id":"3","nozzle_temp_max":"240","nozzle_temp_min":"190","remain":0,"tag_uid":"0000000000000000","tray_color":"161616FF","tray_diameter":"0.00","tray_id_name":"","tray_info_idx":"GFL00","tray_sub_brands":"","tray_type":"PLA","tray_uuid":"00000000000000000000000000000000","tray_weight":"0","xcam_info":"000000000000000000000000"}]},{"humidity":"3","id":"1","temp":"24.6","tray":[{"bed_temp":"0","bed_temp_type":"0","cali_idx":-1,"cols":["898989FF"],"ctype":0,"drying_temp":"0","drying_time":"0","id":"0","nozzle_temp_max":"240","nozzle_temp_min":"190","remain":0,"tag_uid":"0000000000000000","tray_color":"898989FF","tray_diameter":"0.00","tray_id_name":"","tray_info_idx":"GFL99","tray_sub_brands":"","tray_type":"PLA","tray_uuid":"00000000000000000000000000000000","tray_weight":"0","xcam_info":"000000000000000000000000"},{"bed_temp":"0","bed_temp_type":"0","cali_idx":-1,"cols":["161616FF"],"ctype":0,"drying_temp":"0","drying_time":"0","id":"1","nozzle_temp_max":"240","nozzle_temp_min":"190","remain":0,"tag_uid":"0000000000000000","tray_color":"161616FF","tray_diameter":"0.00","tray_id_name":"","tray_info_idx":"GFL99","tray_sub_brands":"","tray_type":"PLA","tray_uuid":"00000000000000000000000000000000","tray_weight":"0","xcam_info":"000000000000000000000000"},{"bed_temp":"0","bed_temp_type":"0","cali_idx":-1,"cols":["FFFFFFFF"],"ctype":0,"drying_temp":"0","drying_time":"0","id":"2","nozzle_temp_max":"240","nozzle_temp_min":"190","remain":0,"tag_uid":"0000000000000000","tray_color":"FFFFFFFF","tray_diameter":"0.00","tray_id_name":"","tray_info_idx":"GFA00","tray_sub_brands":"","tray_type":"PLA","tray_uuid":"00000000000000000000000000000000","tray_weight":"0","xcam_info":"000000000000000000000000"},{"bed_temp":"0","bed_temp_type":"0","cali_idx":-1,"cols":["DCF478FF"],"ctype":0,"drying_temp":"0","drying_time":"0","id":"3","nozzle_temp_max":"230","nozzle_temp_min":"190","remain":0,"tag_uid":"0000000000000000","tray_color":"DCF478FF","tray_diameter":"0.00","tray_id_name":"","tray_info_idx":"GFL99","tray_sub_brands":"","tray_type":"PLA","tray_uuid":"00000000000000000000000000000000","tray_weight":"0","xcam_info":"000000000000000000000000"}]},{"humidity":"3","id":"2","temp":"25.0","tray":[{"id":"0"},{"id":"1"},{"bed_temp":"0","bed_temp_type":"0","cali_idx":-1,"cols":["76D9F4FF"],"ctype":0,"drying_temp":"0","drying_time":"0","id":"2","nozzle_temp_max":"230","nozzle_temp_min":"190","remain":0,"tag_uid":"0000000000000000","tray_color":"76D9F4FF","tray_diameter":"0.00","tray_id_name":"","tray_info_idx":"GFL00","tray_sub_brands":"","tray_type":"PLA","tray_uuid":"00000000000000000000000000000000","tray_weight":"0","xcam_info":"000000000000000000000000"},{"id":"3"}]}],"ams_exist_bits":"7","insert_flag":true,"power_on_flag":false,"tray_exist_bits":"4f8","tray_is_bbl_bits":"4f8","tray_now":"5","tray_pre":"5","tray_read_done_bits":"4f8","tray_reading_bits":"0","tray_tar":"5","version":35857},"ams_rfid_status":2,"ams_status":768,"aux_part_fan":true,"bed_target_temper":55.0,"bed_temper":55.0,"big_fan1_speed":"0","big_fan2_speed":"8","cali_version":0,"chamber_temper":32.0,"command":"push_status","cooling_fan_speed":"0","fail_reason":"0","fan_gear":9830400,"filam_bak":[],"force_upgrade":false,"gcode_file":"/data/Metadata/plate_1.gcode","gcode_file_prepare_percent":"100","gcode_start_time":"1710194036","gcode_state":"RUNNING","heatbreak_fan_speed":"15","hms":[{"attr":201327360,"code":131087}],"home_flag":115991,"hw_switch_state":1,"ipcam":{"ipcam_dev":"1","ipcam_record":"disable","mode_bits":2,"resolution":"1080p","rtsp_url":"rtsps://192.168.19.103/streaming/live/1","timelapse":"disable","tutk_server":"enable"},"layer_num":60,"lifecycle":"product","lights_report":[{"mode":"on","node":"chamber_light"},{"mode":"flashing","node":"work_light"}],"maintain":131075,"mc_percent":25,"mc_print_error_code":"0","mc_print_stage":"2","mc_print_sub_stage":0,"mc_remaining_time":591,"mess_production_state":"active","msg":0,"nozzle_diameter":"0.2","nozzle_target_temper":235.0,"nozzle_temper":235.0,"nozzle_type":"stainless_steel","online":{"ahb":true,"ext":false,"version":17},"print_error":0,"print_gcode_action":0,"print_real_action":0,"print_type":"cloud","profile_id":"57332291","project_id":"58385706","queue_est":0,"queue_number":0,"queue_sts":0,"queue_total":0,"s_obj":[6304,6334],"sdcard":true,"sequence_id":"121786","spd_lvl":2,"spd_mag":100,"stg":[2,14,1],"stg_cur":0,"subtask_id":"115789214","subtask_name":"switchbox","task_id":"114631471","total_layer_num":225,"upgrade_state":{"ahb_new_version_number":"","ams_new_version_number":"00.00.06.40","consistency_request":false,"dis_state":0,"err_code":0,"ext_new_version_number":"","force_upgrade":false,"idx":17,"message":"","module":"null","new_version_state":1,"ota_new_version_number":"01.07.02.00","progress":"0","sequence_id":0,"sn":"<SERIAL>","status":"IDLE"},"upload":{"file_size":0,"finish_size":0,"message":"Good","oss_url":"","progress":0,"sequence_id":"0903","speed":0,"status":"idle","task_id":"","time_remaining":0,"trouble_id":""},"vt_tray":{"bed_temp":"0","bed_temp_type":"0","cali_idx":-1,"cols":["FFFFFFFF"],"ctype":0,"drying_temp":"0","drying_time":"0","id":"254","nozzle_temp_max":"250","nozzle_temp_min":"200","remain":0,"tag_uid":"0000000000000000","tray_color":"FFFFFFFF","tray_diameter":"0.00","tray_id_name":"","tray_info_idx":"GFU99","tray_sub_brands":"","tray_type":"PLA","tray_uuid":"00000000000000000000000000000000","tray_weight":"0","xcam_info":"000000000000000000000000"},"wifi_signal":"-37dBm","xcam":{"allow_skip_parts":false,"buildplate_marker_detector":true,"first_layer_inspector":true,"halt_print_sensitivity":"medium","print_halt":true,"printing_monitor":true,"spaghetti_detector":true},"xcam_status":"0"}}: com.google.protobuf.InvalidProtocolBufferException: Expect message object but got: 6304
bambuweb_1  |   at com.google.protobuf.util.JsonFormat$ParserImpl.mergeMessage(JsonFormat.java:1458)
bambuweb_1  |   at com.google.protobuf.util.JsonFormat$ParserImpl.merge(JsonFormat.java:1435)
bambuweb_1  |   at com.google.protobuf.util.JsonFormat$ParserImpl.parseFieldValue(JsonFormat.java:1995)
bambuweb_1  |   at com.google.protobuf.util.JsonFormat$ParserImpl.mergeRepeatedField(JsonFormat.java:1710)
bambuweb_1  |   at com.google.protobuf.util.JsonFormat$ParserImpl.mergeField(JsonFormat.java:1642)
bambuweb_1  |   at com.google.protobuf.util.JsonFormat$ParserImpl.mergeMessage(JsonFormat.java:1477)
bambuweb_1  |   at com.google.protobuf.util.JsonFormat$ParserImpl.merge(JsonFormat.java:1435)
bambuweb_1  |   at com.google.protobuf.util.JsonFormat$ParserImpl.parseFieldValue(JsonFormat.java:1995)
bambuweb_1  |   at com.google.protobuf.util.JsonFormat$ParserImpl.mergeOneofField(JsonFormat.java:1685)
bambuweb_1  |   at com.google.protobuf.util.JsonFormat$ParserImpl.mergeField(JsonFormat.java:1644)
bambuweb_1  |   at com.google.protobuf.util.JsonFormat$ParserImpl.mergeMessage(JsonFormat.java:1477)
bambuweb_1  |   at com.google.protobuf.util.JsonFormat$ParserImpl.merge(JsonFormat.java:1435)
bambuweb_1  |   at com.google.protobuf.util.JsonFormat$ParserImpl.merge(JsonFormat.java:1317)
bambuweb_1  |   at com.google.protobuf.util.JsonFormat$Parser.merge(JsonFormat.java:449)
bambuweb_1  |   at com.tfyre.bambu.printer.BambuPrinterImpl.fromJson(BambuPrinterImpl.java:197)
bambuweb_1  |   at com.tfyre.bambu.printer.BambuPrinterImpl.process(BambuPrinterImpl.java:221)
bambuweb_1  |   at org.apache.camel.support.processor.DelegateSyncProcessor.process(DelegateSyncProcessor.java:65)
bambuweb_1  |   at org.apache.camel.processor.errorhandler.RedeliveryErrorHandler$SimpleTask.handleFirst(RedeliveryErrorHandler.java:462)
bambuweb_1  |   at org.apache.camel.processor.errorhandler.RedeliveryErrorHandler$SimpleTask.run(RedeliveryErrorHandler.java:438)
bambuweb_1  |   at org.apache.camel.impl.engine.DefaultReactiveExecutor$Worker.doRun(DefaultReactiveExecutor.java:199)
[... cut ...]
bambuweb_1  |   at org.apache.camel.component.paho.PahoConsumer$1.messageArrived(PahoConsumer.java:99)
bambuweb_1  |   at org.eclipse.paho.client.mqttv3.internal.CommsCallback.deliverMessage(CommsCallback.java:519)
bambuweb_1  |   at org.eclipse.paho.client.mqttv3.internal.CommsCallback.handleMessage(CommsCallback.java:417)
bambuweb_1  |   at org.eclipse.paho.client.mqttv3.internal.CommsCallback.run(CommsCallback.java:214)
bambuweb_1  |   at java.base/java.lang.Thread.run(Thread.java:1583)
bambuweb_1  |
`

TFyre, 2024-03-12:
Can you please test with pre-release 1.1.1 https://github.com/TFyre/bambu-farm/releases/tag/v1.1.1 ?

Pedalnetics, 2024-03-12:
Doesn't seem to have changed anything, got the exact same stacktrace (minus the live/changing data in the body). Confirmed I'm on the head @ v1.1.1:
root@bambufarm:/opt/bambu-farm/docker/bambu-liveview# git status
HEAD detached at v1.1.1

TFyre, 2024-03-12:
when you look at the logs for the bambuweb container, it should show the version:
```
2024-03-12 17:36:02,939 INFO  [io.quarkus] (main) bambu-web 1.1.1 on JVM (powered by Quarkus 3.8.2) started in 8.453s. Listening on: http://0.0.0.0:8086

```

Pedalnetics, 2024-03-12:
Ah bingo, that was it; forgot compose needed the manual-jar download/update, first time doing that update =)
No more java stacktrace, I'd say this one is solved.  I'll let you close the ticket when you're ready/with a release/however.

TFyre, 2024-03-12:
Closing the issue as 1.1.1 has been released now 😄

commit 74148b79 (2024-03-12): Implement parsing of unknown message `s_obj` (fixes #50)
closed 2024-03-12 by TFyre