TFyre/bambu-farm #51: Weird dashboard column view

wordym, 2024-03-13:
When viewing dashboard on 1080p monitor with 100% page zoom with two printers only one column is shown. Two columns are shown when zooming below 80%. This makes the 100% zoom webcam field unnecessarily large unless zoomed 80% (which is slightly annoying to do every time).
styles.css file for number of columns seems to have no effect.
[image]
[image]

TFyre, 2024-03-13:
Please share your styles.css

The default is 3 columns on 1080p, so your styles.css does have an impact

wordym, 2024-03-13:
I have tried to run the app without styles.css and now it is working in the three columns view.
If I run it with the styles.css file (attached, just with changed extension so github lets me upload it) it behaves as screenshotted above.
[styles.txt](https://github.com/TFyre/bambu-farm/files/14585546/styles.txt)

TFyre, 2024-03-13:
`--bambu-default-columns` is used as a ratio to calculate columns based on various resolution sizes.

You can see details here:
https://github.com/TFyre/bambu-farm/blob/main/bambu/frontend/themes/bambu-theme/bambu.css#L1-L25

So for 1080pixels it will use the value you have -1... So if you change it to 3 it will display 2 columns

put this in styles.css:
```css
:root {
  --bambu-default-columns: 3;
}
```

wordym, 2024-03-13:
Ah, thank you, this is working. However, as more of a regular user than developer, the text in the "Custom CSS" section might need a bit of a rework for the future, in my opinion - I would expect behavior: number in css = number of columns in dashboard. If this is not easily achievable a comment on the "Custom CSS" paragraph would be nice :)
Anyway thanks for the quick response and software!

closed 2024-03-13 by wordym