TFyre/bambu-farm #55: SD-Card print missing thumbnail and AMS choice

dfjaapie, 2024-03-20:
With the SD-Print function i'm missing a thumbnail and AMS filament choice like you have in the batchprint. This makes it difficult to know what plate to print and impossible to chose the AMS slot.

TFyre, 2024-03-20:
I dont think it makes sense to do this for the SD Card functionality, since it would first have to download the file from the SD card to get the thumbnail & metadata for filament mapping

Why not use the batch print feature for this?

dfjaapie, 2024-03-20:
> I dont think it makes sense to do this for the SD Card functionality, since it would first have to download the file from the SD card to get the thumbnail & metadata for filament mapping
> 
> Why not use the batch print feature for this?

The batch feature is really great when printing new files or printing on multiple printers. But a lot of my models I print from SD-Card because I print them all the time. Using the batch feature for this is not the best choice because the file is already on the printer (batch will issue a new FTP upload) and local file search to the print file).

Aghost917, 2025-04-12:
>But a lot of my models I print from SD-Card because I print them all the time. Using the batch feature for this is not the best choice because the file is already on the printer (batch will issue a new FTP upload) and local file search to the print file).

Check out 1.6.0 new feature: skip upload if file exists and is the same size in batch printing. It is so amazing!