TFyre/bambu-farm #62: Detailed tutorial

Maozaoaoo, 2024-04-01:
I am not very familiar with java compilation, but I am very interested in this project, so there are some detailed tutorials on running this project

TFyre, 2024-04-01:
You dont need to compile it, a compiled version is already provided in the releases section.

Did you check the [I just want to run it](https://github.com/TFyre/bambu-farm?tab=readme-ov-file#i-just-want-to-run-it) section in the readme?

How many of which model printers do you have?

Maozaoaoo, 2024-04-01:
Hi, I just confuse about the .env parts. And I have 5 p1p. Thx

TFyre, 2024-04-01:
You create multiple sections like the `myprinter1`

```properties
bambu.printers.myprinter1.device-id=REPLACE_WITH_DEVICE_SERIAL
bambu.printers.myprinter1.access-code=REPLACE_WITH_DEVICE_ACCESSCODE
bambu.printers.myprinter1.ip=REPLACE_WITH_DEVICE_IP

bambu.printers.myprinter2.device-id=REPLACE_WITH_DEVICE_SERIAL
bambu.printers.myprinter2.access-code=REPLACE_WITH_DEVICE_ACCESSCODE
bambu.printers.myprinter2.ip=REPLACE_WITH_DEVICE_IP

bambu.printers.myprinter3.device-id=REPLACE_WITH_DEVICE_SERIAL
bambu.printers.myprinter3.access-code=REPLACE_WITH_DEVICE_ACCESSCODE
bambu.printers.myprinter3.ip=REPLACE_WITH_DEVICE_IP
```

Maozaoaoo, 2024-04-02:
Hi, I have create the .evn but it does not run correctly.

TFyre, 2024-04-02:
You need to be more specific than that 😄

TFyre, 2024-04-03:
From the readme:
> If starting correctly, it will show `Routes startup (total:10 started:0 disabled:10)` with a number that is 2x your printer count

I dont think the file is called `.env`, its probably something like `env.txt`. please confirm?

from command prompt you can do `dir` to see the files and then `rename OLDNAME.BLAH .env` and restart the app

Maozaoaoo, 2024-04-03:
Thank you very much!

TFyre, 2024-04-03:
Sorted?

tdw3d, 2024-04-26:
I'm having the same issue, but I'm fairly sure I have the env correct:

$ ls
bambu-web-1.1.1-runner.jar  config.env

$ java -jar bambu-web-1.1.1-runner.jar
__  ____  __  _____   ___  __ ____  ______
 --/ __ \/ / / / _ | / _ \/ //_/ / / / __/
 -/ /_/ / /_/ / __ |/ , _/ ,< / /_/ /\ \
--\___\_\____/_/ |_/_/|_/_/|_|\____/___/
2024-04-25 19:39:53,551 INFO  [org.apa.cam.uti.con.ThreadType] (main) The type of thread detected is PLATFORM
2024-04-25 19:39:53,829 INFO  [io.und.servlet] (main) Initializing AtmosphereFramework
2024-04-25 19:39:53,845 INFO  [com.vaa.flo.ser.sta.ServletDeployer] (main) Skipping automatic servlet registration because there is already a Vaadin servlet with the name com.vaadin.quarkus.QuarkusVaadinServlet
2024-04-25 19:39:53,852 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereHandler com.vaadin.flow.server.communication.PushAtmosphereHandler mapped to context-path: /*
2024-04-25 19:39:53,853 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed the following AtmosphereInterceptor mapped to AtmosphereHandler com.vaadin.flow.server.communication.PushAtmosphereHandler
2024-04-25 19:39:53,872 INFO  [org.atm.cpr.AtmosphereFramework] (main) Atmosphere is using org.atmosphere.util.VoidAnnotationProcessor for processing annotation
2024-04-25 19:39:53,875 INFO  [org.atm.uti.ForkJoinPool] (main) Using ForkJoinPool  java.util.concurrent.ForkJoinPool. Set the org.atmosphere.cpr.broadcaster.maxAsyncWriteThreads to -1 to fully use its power.
2024-04-25 19:39:53,879 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed WebSocketProtocol org.atmosphere.websocket.protocol.SimpleHttpProtocol
2024-04-25 19:39:53,881 INFO  [org.atm.con.JSR356AsyncSupport] (main) JSR 356 Mapping path /VAADIN/push
2024-04-25 19:39:53,882 INFO  [io.und.websockets] (main) UT026005: Adding programmatic server endpoint class org.atmosphere.container.JSR356Endpoint for path /VAADIN/push
2024-04-25 19:39:53,883 INFO  [io.und.websockets] (main) UT026005: Adding programmatic server endpoint class org.atmosphere.container.JSR356Endpoint for path /VAADIN/push/
2024-04-25 19:39:53,883 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installing Default AtmosphereInterceptors
2024-04-25 19:39:53,883 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.CorsInterceptor : CORS Interceptor Support
2024-04-25 19:39:53,884 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.CacheHeadersInterceptor : Default Response's Headers Interceptor
2024-04-25 19:39:53,884 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.PaddingAtmosphereInterceptor : Browser Padding Interceptor Support
2024-04-25 19:39:53,884 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.AndroidAtmosphereInterceptor : Android Interceptor Support
2024-04-25 19:39:53,885 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.HeartbeatInterceptor : Heartbeat Interceptor Support
2024-04-25 19:39:53,885 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.SSEAtmosphereInterceptor : SSE Interceptor Support
2024-04-25 19:39:53,885 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.JSONPAtmosphereInterceptor : JSONP Interceptor Support
2024-04-25 19:39:53,886 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.JavaScriptProtocol : Atmosphere JavaScript Protocol
2024-04-25 19:39:53,886 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.WebSocketMessageSuspendInterceptor : org.atmosphere.interceptor.WebSocketMessageSuspendInterceptor
2024-04-25 19:39:53,886 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.OnDisconnectInterceptor : Browser disconnection detection
2024-04-25 19:39:53,886 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.IdleResourceInterceptor : org.atmosphere.interceptor.IdleResourceInterceptor
2024-04-25 19:39:53,886 INFO  [org.atm.cpr.AtmosphereFramework] (main) Set org.atmosphere.cpr.AtmosphereInterceptor.disableDefaults to disable them.
2024-04-25 19:39:53,886 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor CORS Interceptor Support with priority FIRST_BEFORE_DEFAULT
2024-04-25 19:39:53,886 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor Default Response's Headers Interceptor with priority AFTER_DEFAULT
2024-04-25 19:39:53,886 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor Browser Padding Interceptor Support with priority AFTER_DEFAULT
2024-04-25 19:39:53,887 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor Android Interceptor Support with priority AFTER_DEFAULT
2024-04-25 19:39:53,887 INFO  [org.atm.int.HeartbeatInterceptor] (main) HeartbeatInterceptor configured with padding value 'X', client frequency 0 seconds and server frequency 60 seconds
2024-04-25 19:39:53,887 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor Heartbeat Interceptor Support with priority AFTER_DEFAULT

2024-04-25 19:39:53,887 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor SSE Interceptor Support with priority AFTER_DEFAULT
2024-04-25 19:39:53,887 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor JSONP Interceptor Support with priority AFTER_DEFAULT
[... cut ...]
$ 2024-04-25 19:40:17,218 INFO  [org.apa.cam.imp.eng.AbstractCamelContext] (main)                                                                               Apache Camel 4.4.0 (camel-1) is shutting down (timeout:45s)
2024-04-25 19:40:17,220 INFO  [org.apa.cam.imp.eng.AbstractCamelContext] (main) Routes stopped (stopped:0)
2024-04-25 19:40:17,221 INFO  [org.apa.cam.imp.eng.AbstractCamelContext] (main) Apache Camel 4.4.0 (camel-1) shutdown in 3ms (uptime:23s)
2024-04-25 19:40:17,224 INFO  [com.tfy.bam.pri.BambuPrintersImpl] (main) Stopping Printers
2024-04-25 19:40:17,228 INFO  [io.quarkus] (main) bambu-web stopped in 0.025s

TFyre, 2024-04-26:
@tdw3d please rename your config 😄 

```bash
mv config.env .env
```

blasirio, 2024-05-05:
I'm getting something similar

May 04, 2024 10:14:54 PM org.apache.camel.util.concurrent.ThreadType
INFO: The type of thread detected is PLATFORM
May 04, 2024 10:14:54 PM io.undertow.servlet
INFO: Initializing AtmosphereFramework
May 04, 2024 10:14:54 PM com.vaadin.flow.server.startup.ServletDeployer
INFO: Skipping automatic servlet registration because there is already a Vaadin servlet with the name com.vaadin.quarkus.QuarkusVaadinServlet
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO: Installed AtmosphereHandler com.vaadin.flow.server.communication.PushAtmosphereHandler mapped to context-path: /*
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO: Installed the following AtmosphereInterceptor mapped to AtmosphereHandler com.vaadin.flow.server.communication.PushAtmosphereHandler
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO: Atmosphere is using org.atmosphere.util.VoidAnnotationProcessor for processing annotation
May 04, 2024 10:14:54 PM org.atmosphere.util.ForkJoinPool
INFO: Using ForkJoinPool  java.util.concurrent.ForkJoinPool. Set the org.atmosphere.cpr.broadcaster.maxAsyncWriteThreads to -1 to fully use its power.
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO: Installed WebSocketProtocol org.atmosphere.websocket.protocol.SimpleHttpProtocol
May 04, 2024 10:14:54 PM org.atmosphere.container.JSR356AsyncSupport
INFO: JSR 356 Mapping path /VAADIN/push
May 04, 2024 10:14:54 PM io.undertow.websockets
INFO: UT026005: Adding programmatic server endpoint class org.atmosphere.container.JSR356Endpoint for path /VAADIN/push
May 04, 2024 10:14:54 PM io.undertow.websockets
INFO: UT026005: Adding programmatic server endpoint class org.atmosphere.container.JSR356Endpoint for path /VAADIN/push/
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO: Installing Default AtmosphereInterceptors
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO:   org.atmosphere.interceptor.CorsInterceptor : CORS Interceptor Support
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO:   org.atmosphere.interceptor.CacheHeadersInterceptor : Default Response's Headers Interceptor
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO:   org.atmosphere.interceptor.PaddingAtmosphereInterceptor : Browser Padding Interceptor Support
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO:   org.atmosphere.interceptor.AndroidAtmosphereInterceptor : Android Interceptor Support
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO:   org.atmosphere.interceptor.HeartbeatInterceptor : Heartbeat Interceptor Support
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO:   org.atmosphere.interceptor.SSEAtmosphereInterceptor : SSE Interceptor Support
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO:   org.atmosphere.interceptor.JSONPAtmosphereInterceptor : JSONP Interceptor Support
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO:   org.atmosphere.interceptor.JavaScriptProtocol : Atmosphere JavaScript Protocol
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO:   org.atmosphere.interceptor.WebSocketMessageSuspendInterceptor : org.atmosphere.interceptor.WebSocketMessageSuspendInterceptor
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO:   org.atmosphere.interceptor.OnDisconnectInterceptor : Browser disconnection detection
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO:   org.atmosphere.interceptor.IdleResourceInterceptor : org.atmosphere.interceptor.IdleResourceInterceptor
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO: Set org.atmosphere.cpr.AtmosphereInterceptor.disableDefaults to disable them.
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO: Installed AtmosphereInterceptor CORS Interceptor Support with priority FIRST_BEFORE_DEFAULT
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO: Installed AtmosphereInterceptor Default Response's Headers Interceptor with priority AFTER_DEFAULT
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO: Installed AtmosphereInterceptor Browser Padding Interceptor Support with priority AFTER_DEFAULT
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO: Installed AtmosphereInterceptor Android Interceptor Support with priority AFTER_DEFAULT
May 04, 2024 10:14:54 PM org.atmosphere.interceptor.HeartbeatInterceptor
INFO: HeartbeatInterceptor configured with padding value 'X', client frequency 0 seconds and server frequency 60 seconds
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO: Installed AtmosphereInterceptor Heartbeat Interceptor Support with priority AFTER_DEFAULT
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO: Installed AtmosphereInterceptor SSE Interceptor Support with priority AFTER_DEFAULT
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO: Installed AtmosphereInterceptor JSONP Interceptor Support with priority AFTER_DEFAULT
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO: Installed AtmosphereInterceptor Atmosphere JavaScript Protocol with priority AFTER_DEFAULT
May 04, 2024 10:14:54 PM org.atmosphere.cpr.AtmosphereFramework
INFO: Installed AtmosphereInterceptor org.atmosphere.interceptor.WebSocketMessageSuspendInterceptor with priority AFTER_DEFAULT
[... cut ...]
        java.lang.IllegalArgumentException: SRCFG00039: The config property bambu.printers.myprinter2.model with the config value "A1" threw an Exception whilst being converted [A1] cannot be converted to PrinterModel
        at io.smallrye.config.ConfigMappingProvider.mapConfiguration(ConfigMappingProvider.java:75)
        at io.smallrye.config.SmallRyeConfig.<init>(SmallRyeConfig.java:84)
        at io.smallrye.config.SmallRyeConfigBuilder.build(SmallRyeConfigBuilder.java:724)
        at io.quarkus.runtime.generated.Config.readConfig(Unknown Source)
        at io.quarkus.runtime.generated.Config.createRunTimeConfig(Unknown Source)
        ... 8 more

TFyre, 2024-05-05:
@blasirio please check #77

TFyre, 2024-05-22:
Closing issue due to non activity

closed 2024-05-22 by TFyre