TFyre/bambu-farm #70: Simplify installation using Docker / Docker-Compose

RyanEwen, 2024-04-18:
Hi there, 

This project looks amazing, however having a bunch of prerequisites like Java gets in the way for me.

I would like to request the addition of a configured `docker-compose.yml` that would allow someone to get the entire thing running in just a couple of commands. 

Has anyone already done this?

I'm surprised the live-view portion uses docker, but not the rest. 

Cheers

RyanEwen, 2024-04-19:
Aha, I was able to get this going by cherry-picking parts of the liveview docker example (just the bambuweb service).

blasirio, 2024-05-05:
Do you think it will run in Synology Docker?

TFyre, 2024-05-05:
@blasirio Should work if theres enough resources :)

Aghost917, 2024-05-19:
How would I install this on Docker?

TFyre, 2024-05-19:
> How would I install this on Docker?

Check out https://github.com/TFyre/bambu-farm/tree/main/docker/bambu-liveview

TFyre, 2024-05-22:
Closing issue due to non activity

closed 2024-05-22 by TFyre