TFyre/bambu-farm #73: Cannot connect to printers

MatthiasJ1, 2024-04-25:
I have been unable to connect to either of my printers. I can ping them from the computer running bambu-farm and I triple-checked that I specified the correct parameters for each printer in my `.env` file, yet I get the following error:

```
ERROR [com.tfy.bam.pri.BambuPrintersImpl] (executor-thread-1) myprinter1: Unknown Exception: com.tfyre.bambu.printer.BambuPrinterException: myprinter1: Error starting route: consumer-myprinter1: com.tfyre.bambu.printer.BambuPrinterException: Unknown Exception: com.tfyre.bambu.printer.BambuPrinterException: myprinter1: Error starting route: consumer-myprinter1
	at com.tfyre.bambu.printer.BambuPrintersImpl.startPrinter(BambuPrintersImpl.java:117)
	at com.tfyre.bambu.printer.BambuPrintersImpl.stopStart(BambuPrintersImpl.java:174)
	...
Caused by: com.tfyre.bambu.printer.BambuPrinterException: myprinter1: Error starting route: consumer-myprinter1
	at com.tfyre.bambu.printer.BambuPrintersImpl.startPrinter(BambuPrintersImpl.java:107)
	... 14 more
Caused by: org.apache.camel.RuntimeCamelException: Connection lost (32109) - java.io.EOFException
	at org.apache.camel.RuntimeCamelException.wrapRuntimeException(RuntimeCamelException.java:66)
	...
	at org.apache.camel.impl.engine.DefaultRouteController.startRoute(DefaultRouteController.java:133)
	at com.tfyre.bambu.printer.BambuPrintersImpl.startPrinter(BambuPrintersImpl.java:105)
	... 14 more
Caused by: Connection lost (32109) - java.io.EOFException
	at org.eclipse.paho.client.mqttv3.internal.CommsReceiver.run(CommsReceiver.java:197)
	... 1 more
Caused by: java.io.EOFException
	at java.base/java.io.DataInputStream.readUnsignedByte(DataInputStream.java:297)
	at java.base/java.io.DataInputStream.readByte(DataInputStream.java:275)
	at org.eclipse.paho.client.mqttv3.internal.wire.MqttInputStream.readMqttWireMessage(MqttInputStream.java:92)
	at org.eclipse.paho.client.mqttv3.internal.CommsReceiver.run(CommsReceiver.java:137)
	... 1 more
```

my `.env`:
```
quarkus.http.host=0.0.0.0
quarkus.http.port=8080

bambu.printers.myprinter1.device-id=123-456-789
bambu.printers.myprinter1.access-code=12345678
bambu.printers.myprinter1.ip=10.0.0.xxx

bambu.users.admin.password=admin
bambu.users.admin.role=admin
```

I've also tried all combinations of ambiguous characters in the printer serial #, as well as shutting off any other MQTT services running on my network/quitting Bambu Studio.
Do you have any suggestions on how to debug this further?

TFyre, 2024-04-28:
```properties
bambu.printers.myprinter1.device-id=123-456-789
```
Does this have `-`'s in it? if so, remove them to look like this:
```properties
bambu.printers.myprinter1.device-id=123456789
```

MatthiasJ1, 2024-04-29:
I tried without the `-` and I get the same error.

TFyre, 2024-04-29:
The printer closes the mqtt connection... probably due to some non printable characters in the config... please make sure you dont havy any funny characters in there and paste the full stack trace...

Also share the actual config and mask the middle parts

MatthiasJ1, 2024-04-29:
Thanks for looking into this.

I am using the latest `bambu-web-1.1.1-runner.jar` downloaded from the releases page and I am running it with `java -jar bambu-web-1.1.1-runner.jar` in the same directory as the `.env` file.

Java version: `openjdk 22.0.1-internal 2024-04-16`
OS: `Ubuntu 22.04.4 LTS`

I also tried it inside of a docker image running Ubuntu 24 and Java 21 to make sure it wasn't a version issue and I got the same error.

```properties
quarkus.http.host=0.0.0.0
quarkus.http.port=8080

bambu.printers.myprinter1.device-id=3DP01P038
bambu.printers.myprinter1.access-code=42210930
bambu.printers.myprinter1.ip=10.0.0.12

bambu.users.admin.password=admin
bambu.users.admin.role=admin
```

I checked for non printable characters and I didn't notice any but just in case I included the zipped env file: [env.zip](https://github.com/TFyre/bambu-farm/files/15155370/env.zip)

TFyre, 2024-04-30:
1. please include the full stack trace
2. is this masked?
```
bambu.printers.myprinter1.device-id=3DP01P038
```

I was expecting something like (where the X's represent other characters):
```
bambu.printers.myprinter1.device-id=01PXXXXXXXXX230
```

MatthiasJ1, 2024-05-01:
Ah that was the issue! I thought `device-id` was the 9 digit code XXX-XXX-XXX shown under Settings->Device. Thank you for your help, everything works now!

closed 2024-05-01 by MatthiasJ1