TFyre/bambu-farm #97: Unknown Exception upon entering the web interface?

bolsoncerrado, 2024-06-11:
Hi!

First time setting this up, I launch the jar with the minimal .env created with 3 machines (P1P) which ARE CURRENTLY PRINTING (dunno if that makes a difference) but as soon as I log into the web interface and the 3 machines are supposed to be listed, the terminal window is flooded with red text and the following text repeats over and over:

2024-06-11 10:36:38,372 ERROR [com.tfy.bam.cam.CamelController] (executor-thread-1) onCamelContextFullyStarted: Errors with stopStart: [myprinter1: Unknown Exception: com.tfyre.bambu.printer.BambuPrinterException: myprinter1: Error starting route: consumer-myprinter1, myprinter2: Unknown Exception: com.tfyre.bambu.printer.BambuPrinterException: myprinter2: Error starting route: consumer-myprinter2, myprinter3: Unknown Exception: com.tfyre.bambu.printer.BambuPrinterException: myprinter3: Error starting route: consumer-myprinter3]: com.tfyre.bambu.printer.BambuPrinterException: Errors with stopStart: [myprinter1: Unknown Exception: com.tfyre.bambu.printer.BambuPrinterException: myprinter1: Error starting route: consumer-myprinter1, myprinter2: Unknown Exception: com.tfyre.bambu.printer.BambuPrinterException: myprinter2: Error starting route: consumer-myprinter2, myprinter3: Unknown Exception: com.tfyre.bambu.printer.BambuPrinterException: myprinter3: Error starting route: consumer-myprinter3]

What's exactly going on?

I made sure serials, ips and access codes were right :??? What am I missing here?

Thanks for this GREAT idea for FARM CONTROL! Much long awaited!!

MaximeBerduck, 2024-06-12:
Hi !
Can you send your .env configuration file without the IDs and access_code of your printers? 

192.168.xxx.xxx is the IP address of your local private network and 127.0.0.1 is the address that allows access to the ports of your computer on which you have the server. This difference is therefore normal (can also be replaced by localhost:8080).
But is your computer connected to the same network as your printers?

bolsoncerrado, 2024-06-12:
yes it is, Bstudio access them without any issues.

Current .env and printers repeat up to 3 instances just for test purposes. Have a few more printers to add granted it works as it should ;)

`alex@Mac-Studio farm % more .env
# Gobal
bambu.auto-login=true
bambu.remote-view=true
bambu.dashboard.remote-view=true
quarkus.http.limits.max-body-size=30M
bambu.dark-mode=true
bambu.users.admin.password=admin
bambu.users.admin.role=admin

quarkus.http.host=192.168.1.2
quarkus.http.port=8080

bambu.printers.myprinter1.name=TARS
bambu.printers.myprinter1.enabled=true
bambu.printers.myprinter1.device-id=01Sxxxxxxxxxxxxxx
bambu.printers.myprinter1.access-code=xxxxxxxxxxxx
bambu.printers.myprinter1.ip=192.68.1.117
bambu.printers.myprinter1.stream.enable=true
bambu.printers.myprinter1.bed-levelling=true
bambu.printers.myprinter1.flow-calibration=false
bambu.printers.myprinter1.model=p1p`

bolsoncerrado, 2024-06-12:
It takes an (almost) exact minute to find out it times out.

TFyre, 2024-06-12:
You can remove all the defaulted options from your config eg:
```properties
bambu.remote-view=true
bambu.dashboard.remote-view=true

bambu.printers.myprinter1.enabled=true
bambu.printers.myprinter1.flow-calibration=false
bambu.printers.myprinter1.model=p1p
```

Why do you set the http host/port directly? I dont think its required
```properties
quarkus.http.host=192.168.1.2
quarkus.http.port=8080
```

Looks like its not able to connect to the printer's mqtt interface. You can test this from terminal:
```bash
ping 192.68.1.117
openssl s_client -connect 192.68.1.117:8883
```

**I believe the IP is incorrect, it should be `192.168.1.117` (168 vs 68)**

bolsoncerrado, 2024-06-12:
LOOOOOOL

STUPID ME!

OF COURSE IT'S 168!

ROTFL!

TESTING NOW!

ULTRATHANKS FOR This overlook!

bolsoncerrado, 2024-06-12:
And of course, FULLY and AUTOMATICALLY WORKING NOW!

GEEZ!

Blame on me! Shame on me!

Thanks again.

closed 2024-06-12 by bolsoncerrado

TFyre, 2024-06-12:
>Blame on me! Shame on me!

No need, it happens to us all 😃