TFyre/bambu-farm #100: ERROR [com.tfy.bam.pri.BambuPrinterStream] (vert.x-eventloop-thread-1) GAMMA: clientFailure: io.netty.channel.AbstractChannel$AnnotatedConnectException - Connection refused

bolsoncerrado, 2024-06-12:
Hi!

I'm getting the Connection Refused error on the live stream of the video feed of the X1C cameras, all my P1P cameras work, so I guess I must have forgotten a setting somewhere?

The error:

2024-06-12 11:52:59,991 ERROR [com.tfy.bam.pri.BambuPrinterStream] (vert.x-eventloop-thread-1) GAMMA: clientFailure: io.netty.channel.AbstractChannel$AnnotatedConnectException - Connection refused: /192.168.1.114:6000

(same error for each X1C)

The sample .env for X1C definition:

bambu.printers.myprinter11.name=GAMMA
bambu.printers.myprinter11.enabled=true
bambu.printers.myprinter11.device-id=00xxxxxxxxxx
bambu.printers.myprinter11.access-code=xxxxxxxxx
bambu.printers.myprinter11.ip=192.168.1.114
bambu.printers.myprinter11.stream.enable=true
bambu.printers.myprinter11.bed-levelling=true
bambu.printers.myprinter11.flow-calibration=false
bambu.printers.myprinter11.model=x1c

What am I missing?

Thanks!

TFyre, 2024-06-12:
Please check item 2 and 3 in the [Features / Supported Devices](https://github.com/TFyre/bambu-farm?tab=readme-ov-file#features--supported-devices) section.

For the X1C & LiveStream, you need to enable `LAN Mode Liveview` and run the solution via docker (or ffmpeg manually)

bolsoncerrado, 2024-06-12:
Oh wow what a hassle just to get a few frames from the live view! Thanks. :'(

closed 2024-06-12 by bolsoncerrado

tommy4111, 2024-06-14:
Did you get it to work?

bolsoncerrado, 2024-06-14:
Honestly I found it overcomplicated for just a couple I own vs the rest of P1P farm...

May be they want to try if the open X1+ firmware Is easier?