TFyre/bambu-farm #101: Feature Request: jar icon for MacOS?

bolsoncerrado, 2024-06-12:
Hi!

I'm not a programmer so I don't know the limitations of such request but as I understand Java is a "universal" environment and I successfully run the .jar container by just doubleclicking it, I also find it "difficult" to trace if it's running with or without issues, etc. so I don't know if you can somehow detect "if OS = macOS then display icon on tray" or something like that?

Yesterday I downloaded the jar and today I completed setting up the .env so it's now running fine and now instead of running it on Terminal window (console) I just double click the .jar on the macOS GUI and forget (so technically I could set it up to exectue on bootup too) but I find it lacking a way to learn if anything's going wrong or not....

So for example Oracle's Java has a Console/Icon control Panel.... But Azul's has'nt?

[image]

Thanks!

TFyre, 2024-06-12:
Its a terminal application... Its not meant to be run as a Desktop App.. For that you have Bambu Studio / OrcaSlicer / Something else.

Its mean to be run as a service on a server / raspberry pi / service on mac

bolsoncerrado, 2024-06-12:
Fully understandable. So if I run it as a service it will show up just as another process in the running process list and that's all anyway?

TFyre, 2024-06-14:
You need something like this:
https://gist.github.com/cmbankester/a3517bd398475a9e3a48

This is for nginx, so it needs to be customized for bambu farm

When I get a moment, I'll add documentation how to do it for Bambu Farm, unless you want to give it a go and contribute the documentation?

bolsoncerrado, 2024-06-14:
I wish! I'm not that computer literate unfortunately! Still struggling to find the correct combo to hide that row of filament css lol

TFyre, 2025-01-04:
Closing stale issue, please re-open if required

closed 2025-01-04 by TFyre