TFyre/bambu-farm #103: Feature Request: Make STATUS of printers more "visible"

bolsoncerrado, 2024-06-13:
Either (or both) making the background of the title cell or the STATUS cell to different colors (user selectable on settings?) depending on what's happening so we can have a quick look at the farm status when we have many printers to take care of.

A quick photoshop mockup of both options (I think both is more visual):

[image]

TFyre, 2024-06-14:
heheheh my kinda design!! many other users would call it tooo colourful!

https://github.com/TFyre/bambu-farm/blob/main/bambu/src/main/java/com/tfyre/bambu/printer/BambuConst.java#L307

When I get a moment, I'll add the GCodeState to the html element so that you can use CSS to target it and set the colour.

bolsoncerrado, 2024-06-14:
Awesome ;) If u can provide the sample CSS, twice awesome haha :P

PR #113 (2024-06-17): Design/show color status
Closes #103 

Easy way to display the printer status in colour. 
- PRIMARY (blue) -> isPrinting
- ERROR (red) -> isError (no change)
- SUCCESS (green) -> isReady

[image]
closed 2024-06-17 by TFyre