TFyre/bambu-farm #104: Feature Request: Make the FILAMENT CHOICE row user hiddeable on Dashboard view.

bolsoncerrado, 2024-06-13:
I do run a farm with single filament printers and I dont have a real use for the space that the row of filament type and color takes on the DASHBOARD view. Can it be hidden somehow via  a user setting please? ;)

Mockup:

[image]

TFyre, 2024-06-14:
Please checkout the [Custom CSS](https://github.com/TFyre/bambu-farm#custom-css) section 😄 

You can use CSS to hide it.

bolsoncerrado, 2024-06-14:
Agreed! Always more desireable to make it user friendly but ok! ;)

TFyre, 2025-01-04:
Adding this to `styles.css` does the trick:

```css
.dashboard-printer div.filaments {
    display: none;
}
```

closed 2025-01-04 by TFyre