TFyre/bambu-farm #105: Development doc

tommy4111, 2024-06-14:
Any docs on architecture and development workflow?

TFyre, 2024-06-14:
I need to add some documentation.

# Architecture

Its a java maven project using quarkus framework. Im using NetBeans IDE, but IntelliJ / VSCode would work without issues too.

Check out https://quarkus.io/guides/getting-started#running-the-application

Build everything using something like `mvn clean install`

# Dev with printer
To run it from the IDE open the `bambu-web` module and add `quarkus:dev` as a maven target to get something like `mvn clean quarkus:dev`

# Dev without printer
For local dev without access to a printer you need to setup some manual stuff:
1) Start mqtt & ftp servers via docker (refer https://github.com/TFyre/bambu-farm/tree/main/docker/bambu-local-dev)
2) open the `bambu-server` module and start the same as `bambu-web`

example `.env` config for `bambu-server`
```properties
bambu.printers.tfyre1.device-id=printer1
bambu.printers.tfyre1.url=ssl://127.0.0.1:8883
bambu.printers.tfyre1.username=test1
bambu.printers.tfyre1.access-code=test1

bambu.printers.tfyre2.device-id=printer2
bambu.printers.tfyre2.url=ssl://127.0.0.1:8883
bambu.printers.tfyre2.username=test2
bambu.printers.tfyre2.access-code=test2

bambu.printers.tfyre3.device-id=printer3
bambu.printers.tfyre3.url=ssl://127.0.0.1:8883
bambu.printers.tfyre3.username=test3
bambu.printers.tfyre3.access-code=test3

bambu.printers.tfyre4.device-id=printer4
bambu.printers.tfyre4.url=ssl://127.0.0.1:8883
bambu.printers.tfyre4.username=test4
bambu.printers.tfyre4.access-code=test4
```

It will check the [resources ](https://github.com/TFyre/bambu-farm/tree/main/server/src/main/resources/json)folder for `fullstatus-tfyreX.json` and then `fullstatus.json` to "randomize" the printers responses a bit, refer to: 
https://github.com/TFyre/bambu-farm/blob/main/server/src/main/java/com/tfyre/bambu/server/BambuPrinterProcessor.java#L93

tommy4111, 2024-06-20:
For dev with a printer, do I run all the modules (bambu-common, bambu-server, bambu-web)?  And where should bambu-web-env.txt file be?

TFyre, 2024-06-20:
bambu-common is a shared library and doesnt need to be run

run bambu-server first and then bambu-web

`.env` files (no filename, only extension) should be in bambu-server and bambu-web (different files though)
```bash
[root@ares:~/bambu-farm]# ls -al
total 364976
drwxr-xr-x.  2 root root     4096 Mar 12 17:35 .
dr-xr-x---. 12 root root     4096 Jun 19 15:29 ..
-rw-r--r--.  1 root root      355 Feb 11 10:47 .env
-rw-r--r--.  1 root root 74746833 Feb  2 16:48 bambu-web-1.0.6-runner.jar
-rw-r--r--.  1 root root 75582905 Feb  9 15:45 bambu-web-1.0.7-runner.jar
-rw-r--r--.  1 root root 74340361 Feb 17 10:57 bambu-web-1.0.8-runner.jar
-rw-r--r--.  1 root root 74515583 Mar  5 18:21 bambu-web-1.1.0-runner.jar
-rw-r--r--.  1 root root 74525526 Mar 12 17:35 bambu-web-1.1.1-runner.jar
-rw-r--r--.  1 root root       83 Feb  1 22:35 styles.css
```

bambu-web-env.txt is only used when you run it in a more production environment with docker compose (for the X1C)

closed 2024-06-21 by tommy4111