TFyre/bambu-farm #106: Upgrade Function

hechay, 2024-06-14:
Hello, 

Your project is very wonderful and I really love it. I just think that you may add a function that can upgrade the firmware for region-limited device that only work in LAN-MODE. 

The newest firmware information may be update here: https://github.com/lunDreame/user-bambulab-firmware/tree/main

Just an idea and may people love it too.

Anyway, thanks for your project again!

TFyre, 2024-06-14:
Sounds like a great idea, but unfortunately I dont have a device to properly test this... Doing firmware upgrades sounds a bit risky? Can you downgrade again if something goes wrong?

hechay, 2024-06-14:
I havent test for downgrading system. However, you can make function to
make custom mqtt request to the printer ( url/data )  that user can do it.
You shouldnt care about the system upgrading. User will do it by themself
and take risk when doing it. It takes me easily that I can upgrade system
without using mqtt software on computer!

hechay, 2024-06-14:
[image]
Make a page that send custom mqtt request to the printer. You may leave the url and the data blank.

hechay, 2024-06-14:
[image]

TFyre, 2025-01-04:
Closing stale issue, please re-open if required

closed 2025-01-04 by TFyre