TFyre/bambu-farm #107: Live view docker compose up error

tommy4111, 2024-06-14:
bambu-liveview-myprinter1-1  | [tcp @ 0x5592ac9aee40] Connection to tcp://192.168.50.138:322?timeout=30000000 failed: Connection refused
bambu-liveview-myprinter1-1  | [in#0 @ 0x5592ac9abd40] Error opening input: Connection refused
bambu-liveview-myprinter1-1  | Error opening input file rtsps://bblp:9fb060c3@192.168.50.138:322/streaming/live/1.
bambu-liveview-myprinter1-1  | Error opening input files: Connection refused
Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting "/run/desktop/mnt/host/c/Users/chris/Downloads/bambu-farm-main/bambu-farm-main/docker/bambu-liveview/bambu-web-1.2.0-runner.jar" to rootfs at "/bambu-web-runner.jar": mount /run/desktop/mnt/host/c/Users/chris/Downloads/bambu-farm-main/bambu-farm-main/docker/bambu-liveview/bambu-web-1.2.0-runner.jar:/bambu-web-runner.jar (via /proc/self/fd/9), flags: 0x5000: not a directory: unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type

closed 2024-06-14 by tommy4111

TFyre, 2024-06-14:
Please add a resolution so that future users can benefit from the opened issue, thx 😄

tommy4111, 2024-06-15:
This one was weird.  I just kept retrying docker-compose up and finally worked.  Maybe old instance still running?