TFyre/bambu-farm #108: Invalid or corrupt jarfile bambu-web-runner.jar

tommy4111, 2024-06-14:
bambu-liveview-myprinter1-1  | [tcp @ 0x55eb5d185e40] Connection to tcp://192.168.50.138:322?timeout=30000000 failed: Connection refused
bambu-liveview-myprinter1-1  | [in#0 @ 0x55eb5d182d40] Error opening input: Connection refused
bambu-liveview-myprinter1-1  | Error opening input file rtsps://bblp:9fb060c3@192.168.50.138:322/streaming/live/1.
bambu-liveview-myprinter1-1  | Error opening input files: Connection refused
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1    | Error: Invalid or corrupt jarfile bambu-web-runner.jar
bambu-liveview-bambuweb-1 exited with code 1

closed 2024-06-14 by tommy4111

TFyre, 2024-06-14:
Please add a resolution so that future users can benefit from the opened issue, thx 😄

tommy4111, 2024-06-15:
I just re-download and copy the bambu-web-1.2.0-runner.jar into bambu-liveview folder again.