TFyre/bambu-farm #112: Get Camera working

tommy4111, 2024-06-15:
Anyone got camera working?  I'm running on Windows.

Went through everything in https://github.com/TFyre/bambu-farm/tree/main/docker/bambu-liveview

Not sure what I'm missing.

Some errors I'm seeing trying various ports:

PRINTER_PORT: 322
```
bambu-liveview-myprinter2-1  | [in#0 @ 0x5585d1825d40] Error opening input: Invalid data found when processing input
bambu-liveview-myprinter2-1  | Error opening input file rtsps://bblp:<KEY>@192.168.50.203:8883/streaming/live/1.
bambu-liveview-myprinter2-1  | Error opening input files: Invalid data found when processing input
bambu-liveview-mediamtx-1    | 2024/06/15 11:26:01 INF [WebRTC] [session 409fd8ce] created by 172.19.0.6:39516
bambu-liveview-mediamtx-1    | 2024/06/15 11:26:01 INF [WebRTC] [session 409fd8ce] closed: no one is publishing to path
```

PRINTER_PORT: 8883
```
 2024/06/15 11:26:44 INF [WebRTC] [session f911256e] closed: no one is publishing to path 'myprinter2'
```

TFyre, 2024-06-15:
As per previous comment https://github.com/TFyre/bambu-farm/issues/111#issuecomment-2169164564, please dont fiddle with the ports unless you are doing fancy port forwarding!

Assuming this is a X1C?

https://github.com/TFyre/bambu-farm/tree/main/docker/bambu-liveview#docker-compose-environment-for-x1c-printers
> REMEMBER to enable LAN Mode Liveview on the printer

Test if you can connect to SSL Video Stream (and paste the output after stripping sensitive information like printer serial):
```bash
openssl s_client -connect 192.168.50.203:322
```

tommy4111, 2024-06-16:
Thanks for reply.  Looks like SSL video stream doesn't work.  Any suggestion?

```
OpenSSL> s_client -connect 192.168.50.203:322
CONNECTED(00000004)
Can't use SSL_get_servername
depth=0 CN = 00M09A391300170
verify error:num=20:unable to get local issuer certificate
verify return:1
depth=0 CN = 00M09A391300170
verify error:num=21:unable to verify the first certificate
verify return:1
depth=0 CN = 00M09A391300170
verify return:1
---
```

tommy4111, 2024-06-17:
Thanks for reply.  Here you go:

```
openssl s_client -connect 192.168.50.203:322
CONNECTED(00000004)
Can't use SSL_get_servername
verify error:num=20:unable to get local issuer certificate
verify return:1
verify error:num=21:unable to verify the first certificate
verify return:1
verify return:1
---
Certificate chain
   i:C = CN, O = "BBL Technologies Co., Ltd", CN = BBL CA
---
Server certificate
-----BEGIN CERTIFICATE-----
MIIC2DCCAcACCQCEvYN7v7TMvjANBgkqhkiG9w0BAQsFADBCMQswCQYDVQQGEwJD
TjEiMCAGA1UECgwZQkJMIFRlY2hub2xvZ2llcyBDby4sIEx0ZDEPMA0GA1UEAwwG
...
ymP1BzOjcdbXR8p9
-----END CERTIFICATE-----
subject=CN = 

issuer=C = CN, O = "BBL Technologies Co., Ltd", CN = BBL CA

---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: RSA
Server Temp Key: ECDH, P-256, 256 bits
---
SSL handshake has read 1386 bytes and written 419 bytes
Verification error: unable to verify the first certificate
---
New, TLSv1.2, Cipher is ECDHE-RSA-AES256
Server public key is 2048 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-RSA-AES256
    Session-ID: E525B68FEE40DBD3B2C6CB01D2C3A5F
    Session-ID-ctx:
    Master-Key: 2CA52D0E3CB3D8E0F194EEB71D88A1760E330C624E41
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 300 (seconds)
    TLS session ticket:
    0000 - 87 aa b3 77 90 37 2d 9d-b6 38 75 bd a2 eb 35 73   ...w.7-..8u...5s

    Start Time: 1718582405
    Timeout   : 7200 (sec)
    Verify return code: 21 (unable to verify the first certificate)
    Extended master secret: no
---
```

TFyre, 2024-06-17:
Awesome!

Please remember liveview (high FPS) is only applicable to X1C. The A1 will use the normal jpg streaming (low FPS) route

tommy4111, 2024-06-17:
Sorry, so base on this certificate error output, I can't make SSL connection right?  Anything else you think I can try?

tommy4111, 2024-06-19:
I'm either getting "invalid data found" or "Error opening input: Connection timed out"

Everything else work, just camera not working.  

```
bambu-liveview-myprinter1-1  | [in#0 @ 0x56412f7b2d40] Error opening input: Invalid data found when processing input
bambu-liveview-myprinter1-1  | Error opening input file rtsps://bblp:9fb060c3@192.168.50.138:8883/streaming/live/1.
bambu-liveview-myprinter1-1  | Error opening input files: Invalid data found when processing input
bambu-liveview-myprinter1-1 exited with code 183
bambu-liveview-myprinter2-1  | [in#0 @ 0x564be6777d40] Error opening input: Invalid data found when processing input
bambu-liveview-myprinter2-1  | Error opening input file rtsps://bblp:2f5716b6@192.168.50.203:8883/streaming/live/1.
bambu-liveview-myprinter2-1  | Error opening input files: Invalid data found when processing input
bambu-liveview-myprinter2-1 exited with code 183
bambu-liveview-myprinter3-1  | [in#0 @ 0x55b04dba9d40] Error opening input: Invalid data found when processing input
bambu-liveview-myprinter3-1  | Error opening input file rtsps://bblp:24516464@192.168.50.68:8883/streaming/live/1.
bambu-liveview-myprinter3-1  | Error opening input files: Invalid data found when processing input
```

```
bambu-liveview-myprinter3-1  | [tcp @ 0x5619d7724e40] Connection to tcp://192.168.50.68:322?timeout=30000000 failed: Connection refused
bambu-liveview-myprinter3-1  | [in#0 @ 0x5619d7721d40] Error opening input: Connection refused
bambu-liveview-myprinter3-1  | Error opening input file rtsps://bblp:24516464@192.168.50.68:322/streaming/live/1.
bambu-liveview-myprinter3-1  | Error opening input files: Connection refused
```

TFyre, 2024-06-19:
TCP 8883 <--- this is WRONG
TCP 322 <--- this is CORRECT

Please specify the models for myprinter1/2/3

**The streaming is ONLY for x1c.**

for the A1, you need to use the normal jpeg frames

tommy4111, 2024-06-20:
Got it.  Thank you so much for your help.  I gave up on XC1 for now (only need A1) and A1 is working.

closed 2024-06-20 by tommy4111

TFyre, 2024-06-20:
Dont give up on the X1C 😄 

Below is GOOD!!!
```
openssl s_client -connect 192.168.50.203:322
CONNECTED(00000004)
```

Below is where it goes wrong... Change the port from `8883` to `322` and it should work
```
bambu-liveview-myprinter2-1  | [in#0 @ 0x564be6777d40] Error opening input: Invalid data found when processing input
bambu-liveview-myprinter2-1  | Error opening input file rtsps://bblp:2f5716b6@192.168.50.203:8883/streaming/live/1.
bambu-liveview-myprinter2-1  | Error opening input files: Invalid data found when processing input
bambu-liveview-myprinter2-1 exited with code 183
```