TFyre/bambu-farm #122: Camera only screen

bonninjd, 2024-06-24:
Great software.  It would be great to have a camera only screen, so that it can be put on a TV to monitor many printers without scrolling.  maybe the ability to choose how many are in the onscreen grid or auto size the video to have all printers on a single screen.

TFyre, 2024-06-24:
Please check the section in the readme about custom css... you can control the size of everything

bonninjd, 2024-06-24:
I did see that, but didn't see a way to have the cameras only.  I was thinking it would be a tv that hung above the printers and would only have the cameras and id, maybe a status word (running, paused, offline), but no controls.  Strictly visual only.