TFyre/bambu-farm #137: MQTT: Not authorized to connect

ualinker, 2024-08-02:
I have minimal cfg in `.env` file with correct IP, SN and access code. Unfortunately, this is what I got while I'm trying to start the server:

```
 $ java -Djdk.tls.useExtendedMasterSecret=false -jar bambu-web-1.2.1-runner.jar
__  ____  __  _____   ___  __ ____  ______
 --/ __ \/ / / / _ | / _ \/ //_/ / / / __/
 -/ /_/ / /_/ / __ |/ , _/ ,< / /_/ /\ \
--\___\_\____/_/ |_/_/|_/_/|_|\____/___/
2024-08-02 23:19:40,894 INFO  [io.und.servlet] (main) Initializing AtmosphereFramework
2024-08-02 23:19:40,902 INFO  [com.vaa.flo.ser.sta.ServletDeployer] (main) Skipping automatic servlet registration because there is already a Vaadin servlet with the name com.vaadin.quarkus.QuarkusVaadinServlet
2024-08-02 23:19:40,910 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereHandler com.vaadin.flow.server.communication.PushAtmosphereHandler mapped to context-path: /*
2024-08-02 23:19:40,910 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed the following AtmosphereInterceptor mapped to AtmosphereHandler com.vaadin.flow.server.communication.PushAtmosphereHandler
2024-08-02 23:19:40,926 INFO  [org.atm.cpr.AtmosphereFramework] (main) Atmosphere is using org.atmosphere.util.VoidAnnotationProcessor for processing annotation
2024-08-02 23:19:40,929 INFO  [org.atm.uti.ForkJoinPool] (main) Using ForkJoinPool  java.util.concurrent.ForkJoinPool. Set the org.atmosphere.cpr.broadcaster.maxAsyncWriteThreads to -1 to fully use its power.
2024-08-02 23:19:40,933 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed WebSocketProtocol org.atmosphere.websocket.protocol.SimpleHttpProtocol
2024-08-02 23:19:40,934 INFO  [org.atm.con.JSR356AsyncSupport] (main) JSR 356 Mapping path /VAADIN/push
2024-08-02 23:19:40,936 INFO  [io.und.websockets] (main) UT026005: Adding programmatic server endpoint class org.atmosphere.container.JSR356Endpoint for path /VAADIN/push
2024-08-02 23:19:40,937 INFO  [io.und.websockets] (main) UT026005: Adding programmatic server endpoint class org.atmosphere.container.JSR356Endpoint for path /VAADIN/push/
2024-08-02 23:19:40,937 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installing Default AtmosphereInterceptors
2024-08-02 23:19:40,937 INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.CorsInterceptor : CORS Interceptor Support
2024-08-02 23:19:40,937 INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.CacheHeadersInterceptor : Default Response's Headers Interceptor
2024-08-02 23:19:40,937 INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.PaddingAtmosphereInterceptor : Browser Padding Interceptor Support
2024-08-02 23:19:40,937 INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.AndroidAtmosphereInterceptor : Android Interceptor Support
2024-08-02 23:19:40,938 INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.HeartbeatInterceptor : Heartbeat Interceptor Support
2024-08-02 23:19:40,938 INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.SSEAtmosphereInterceptor : SSE Interceptor Support
2024-08-02 23:19:40,938 INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.JSONPAtmosphereInterceptor : JSONP Interceptor Support
2024-08-02 23:19:40,938 INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.JavaScriptProtocol : Atmosphere JavaScript Protocol
2024-08-02 23:19:40,938 INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.WebSocketMessageSuspendInterceptor : org.atmosphere.interceptor.WebSocketMessageSuspendInterceptor
2024-08-02 23:19:40,938 INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.OnDisconnectInterceptor : Browser disconnection detection
2024-08-02 23:19:40,938 INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.IdleResourceInterceptor : org.atmosphere.interceptor.IdleResourceInterceptor
2024-08-02 23:19:40,938 INFO  [org.atm.cpr.AtmosphereFramework] (main) Set org.atmosphere.cpr.AtmosphereInterceptor.disableDefaults to disable them.
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor CORS Interceptor Support with priority FIRST_BEFORE_DEFAULT
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor Default Response's Headers Interceptor with priority AFTER_DEFAULT
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor Browser Padding Interceptor Support with priority AFTER_DEFAULT
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor Android Interceptor Support with priority AFTER_DEFAULT
2024-08-02 23:19:40,939 INFO  [org.atm.int.HeartbeatInterceptor] (main) HeartbeatInterceptor configured with padding value 'X', client frequency 0 seconds and server frequency 60 seconds
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor Heartbeat Interceptor Support with priority AFTER_DEFAULT
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor SSE Interceptor Support with priority AFTER_DEFAULT
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor JSONP Interceptor Support with priority AFTER_DEFAULT
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor Atmosphere JavaScript Protocol with priority AFTER_DEFAULT
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor org.atmosphere.interceptor.WebSocketMessageSuspendInterceptor with priority AFTER_DEFAULT
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor Browser disconnection detection with priority AFTER_DEFAULT
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor org.atmosphere.interceptor.IdleResourceInterceptor with priority BEFORE_DEFAULT
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Using EndpointMapper class org.atmosphere.util.DefaultEndpointMapper
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Using BroadcasterCache: org.atmosphere.cache.UUIDBroadcasterCache
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Default Broadcaster Class: org.atmosphere.cpr.DefaultBroadcaster
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Broadcaster Shared List Resources: false
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Broadcaster Polling Wait Time 100
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Shared ExecutorService supported: true
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Messaging ExecutorService Pool Size unavailable - Not instance of ThreadPoolExecutor
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Async I/O Thread Pool Size: 200
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Using BroadcasterFactory: org.atmosphere.cpr.DefaultBroadcasterFactory
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Using AtmosphereResurceFactory: org.atmosphere.cpr.DefaultAtmosphereResourceFactory
2024-08-02 23:19:40,939 INFO  [org.atm.cpr.AtmosphereFramework] (main) Using WebSocketProcessor: org.atmosphere.websocket.DefaultWebSocketProcessor
2024-08-02 23:19:40,940 INFO  [org.atm.cpr.AtmosphereFramework] (main) Invoke AtmosphereInterceptor on WebSocket message true
2024-08-02 23:19:40,940 INFO  [org.atm.cpr.AtmosphereFramework] (main) HttpSession supported: true
2024-08-02 23:19:40,940 INFO  [org.atm.cpr.AtmosphereFramework] (main) Atmosphere is using org.atmosphere.inject.InjectableObjectFactory for dependency injection and object creation
2024-08-02 23:19:40,940 INFO  [org.atm.cpr.AtmosphereFramework] (main) Atmosphere is using async support: org.atmosphere.container.JSR356AsyncSupport running under container: Undertow - 5.2.2.Final using jakarta.servlet/3.0 and jsr356/WebSocket API
2024-08-02 23:19:40,940 INFO  [org.atm.cpr.AtmosphereFramework] (main) Atmosphere Framework 3.0.5.slf4jvaadin1 started.
2024-08-02 23:19:40,941 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor  Track Message Size Interceptor using | with priority BEFORE_DEFAULT
[... cut ...]
	at org.jboss.threads.EnhancedQueueExecutor$ThreadBody.run(EnhancedQueueExecutor.java:1521)
	at org.jboss.threads.DelegatingRunnable.run(DelegatingRunnable.java:11)
	at org.jboss.threads.ThreadLocalResettingRunnable.run(ThreadLocalResettingRunnable.java:11)
	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	at java.base/java.lang.Thread.run(Thread.java:1583)
```

I am able to connect to the printer's IP on port 8883 with `telnet`. Any suggestion how else I can debug this?

TFyre, 2024-08-07:
Please ensure you are using the correct `access-code`

TFyre, 2024-09-07:
Closing due to inactivity

closed 2024-09-07 by TFyre