TFyre/bambu-farm #142: Unable to start print

rontxhou, 2024-08-17:
Thanks for this cool app! I can certainly use this in our farm.

Installed and all controls work fine and can connect to multiple p1p printers. Can upload .3mf files to printer. No AMS units on printers, just external spools.

I am unable to start job via print from SD or batch print. Filament mapped.

Can you please point me where I should be looking for troubleshooting? I do not see any errors on server startup or during the execution of jobs in console output or logs. It says the batch job is sent, but nothing happens with the batch print function or the print from sd card.

rontxhou, 2024-08-18:
Resolved

closed 2024-08-18 by rontxhou

TFyre, 2024-08-19:
How did you fix it? It might help other users with similar issues 😄

rontxhou, 2024-08-19:
Sorry...

I had to downgrade my firmware. In a fairly recent firmware update, I believe that Bamabu blocked MQTT requests with P1Ps not in LAN Only mode. Once I downgraded, it worked fine. 

https://forum.bambulab.com/t/mqtt-for-a1/50033/22

TFyre, 2024-08-19:
Thanks for confirming!

It correllates with their release notes:
https://wiki.bambulab.com/en/p1/manual/p1p-firmware-release-history

> Prevent conflicts with Bambu Cloud service & potential damage to printer hardware by prohibiting crucial printer controls (initiating printing, heating the nozzle, and heating the heatbed) through the local MQTT Broker while the printer is logged into Bambu Cloud service.