TFyre/bambu-farm #148: Help me pls

Dangoreh, 2024-08-29:
Hello, i have problem, i dont understand. why build this project. why create file .env and building?)

TFyre, 2024-08-29:
You need to be a little bit more specific 😄

Dangoreh, 2024-08-30:
> Вам нужно быть немного более конкретным 😄

See. I create everything according to the instructions. Does not allow you to log in to the panel, writes that the admin/ admin account is not correct. I don't understand where I could have made a mistake

Dangoreh, 2024-08-30:
> Вам нужно быть немного более действенным 😄

I'm sorry, I put the question the wrong way) My English is bad)

TFyre, 2024-09-07:
Please share some screenshots / terminal output

SoftwarePixel, 2024-10-29:
@Dangoreh The file must be in the local directory, and named only ".env". If you named it something like "config.env" that will not work!

TFyre, 2025-01-04:
Closing stale issue, please re-open if required

closed 2025-01-04 by TFyre