TFyre/bambu-farm #151: Error: stream not found -  no one is publishing to path

sealltech, 2024-09-04:
I have tried all I can think of to get the video stream to output. The docker images build and the webpage works fine. I have confirmed that it does get a certificate from the output below on 322. I have followed the readme as close as possible. Can you take a look at the diagnostic data below and see if you see something wrong.

Bambu X1C Printer on latest firmware

-----------------------------------------------------------------------------------------------------------------------------------------------
Cert Diagnostic on 322 from command line:
openssl s_client -connect 192.168.1.47:322 -CApath /etc/ssl/certs/
CONNECTED(00000003)
Can't use SSL_get_servername
depth=0 CN = 00M09XXXXXXXXX
verify error:num=20:unable to get local issuer certificate
verify return:1
depth=0 CN =00M09XXXXXXXXX
verify error:num=21:unable to verify the first certificate
verify return:1
depth=0 CN = 00M09XXXXXXXXX
verify return:1
---
Certificate chain
 0 s:CN = 00M09XXXXXXXXX
   i:C = CN, O = "BBL Technologies Co., Ltd", CN = BBL CA
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: May 24 15:13:23 2024 GMT; NotAfter: May 22 15:13:23 2034 GMT
---
Server certificate
-----BEGIN CERTIFICATE-----
MIIC4zCCAcsCFDTUPLWw9smeSZGYwBYfvOTiI2XxMA0GCSqGSIb3DQEBCwUAMEIx
CzAJBgNVBAYTAkNOMSIwIAYDVQQKDBlCQkwgVGVjaG5vbG9naWVzIENvLiwgTHRk
MQ8wDQYDVQQDDAZCQkwgQ0EwHhcNMjQwNTI0MTUxMzIzWhcNMzQwNTIyMTUxMzIz
WjAaMRgwFgYDVQQDDA8wME0wOUM0NTIyMDIxMzIwggEiMA0GCSqGSIb3DQEBAQUA
A4IBDwAwggEKAoIBAQDTAMYu+ZK53NtwIPk/mcnkQSoSmWdKEa/LST8WjF+YvpIg
EGXHUVvsmCUl6FG39WKVzjUCo6yzJpHjtNTipOG9I1dPBrcBCL9t2twLsJ5dW2pa
JjGWKhAd3PQEcHGshFQh34Nv9QNysmIVRXNYE1upQZSfUOBwskgVelHGhI984Qb5
j/JMhImJFWT+vNGvsaE3YLOMApHc8zVU+lTqhcEJl1uyy1Trv7mT4fA++3PsCtY0
dV+jGpT7ppwNxKlhMUXSGBgcSdMIJuYipW/cP6lF9khqVMWQMHj4bTzNWWreAYI6
lhb4o6drVXc6mDiLvWOy2Ax7++jB0khBlhn1/f7lAgMBAAEwDQYJKoZIhvcNAQEL
BQADggEBAMc2bdi0l8aVJdPFwn1gszQmuzAzODxRS9eQyJauQkKJk2OTYN8+4SrG
IvCuLW7lnyKVwsvQzE2HgmU2ePvUrlXK1EoX7l/blrEhTV4PKsQZ8mi46+pePCFT
ssyrcBAvt+vJR5+5C790403OQHbAcWNzFrx0lbAhH7gnyGyM3Zxw5ctshj8qRtJi
OReq+ZKYm8jDwvkIXExoP/b8jCzjPZM+2d4lC6K6FMHL/VndEyRSZHPQRMfraAyE
IDDCow06nejAI28bpUp3iAOW6XMSNdVw+5pK9izpPEtcCJHP2kaZhqorB/NoLcgn
Mi8Y1SEtx5/tLmtII2nqMjEVrbfkk3E=
-----END CERTIFICATE-----
subject=CN = 00M09XXXXXXXXX
issuer=C = CN, O = "BBL Technologies Co., Ltd", CN = BBL CA
---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: RSA
Server Temp Key: ECDH, prime256v1, 256 bits
---
SSL handshake has read 1397 bytes and written 419 bytes
Verification error: unable to verify the first certificate
---
New, TLSv1.2, Cipher is ECDHE-RSA-AES256-GCM-SHA384
Server public key is 2048 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-RSA-AES256-GCM-SHA384
    Session-ID: 303D8F24AF52E297D7CF2ADD7CED314C062806982CA0373ED8E8E04EFABD350F
    Session-ID-ctx:
    Master-Key: 87746D4A45F3AA267BED75F92FDE29BC734E03ABFD3836FC683F5FA18571591E3B075AE9E528F469B4972E07B47EF2C2
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 7200 (seconds)
[... 232 lines cut ...]
bambu-liveview-mediamtx-1  | 2024/09/04 23:34:28 INF [WebRTC] [session 470962f7] created by 172.18.0.5:57484
bambu-liveview-mediamtx-1  | 2024/09/04 23:34:28 INF [WebRTC] [session 470962f7] closed: no one is publishing to path 'r2d2'
bambu-liveview-printer1-1  | size=N/A time=00:00:31.01 bitrate=N/A speed=1.02x
bambu-liveview-printer1-1  | size=N/A time=00:00:31.54 bitrate=N/A speed=1.02x
bambu-liveview-printer1-1  | size=N/A time=00:00:32.04 bitrate=N/A speed=1.02x
bambu-liveview-printer1-1  | size=N/A time=00:00:32.54 bitrate=N/A speed=1.02x

-------------------------------------------------------------------------------------------------------------------------------------------

Config env.txt config Note sensitive items have been obfuscated:
bambu.use-bouncy-castle=true
bambu.live-view-url=/_camerastream/

bambu.users.admin.password=XXXXXXXX
bambu.users.admin.role=admin

bambu.printers.R2D2.stream.live-view=true
bambu.printers.R2D2.enabled=true
bambu.printers.R2D2.name=Roboplop
bambu.printers.R2D2.device-id=XXXXXXXX
bambu.printers.R2D2.username=bblp
bambu.printers.R2D2.access-code=XXXXXXX
bambu.printers.R2D2.ip=192.168.1.XX
bambu.printers.R2D2.use-ams=true
bambu.printers.R2D2.timelapse=true

sealltech, 2024-09-05:
Also for this varible webrtcAdditionalHosts: X.X.X.X is it a single IP needed for this to work or mutiple? Is it the docker host IP its looking for or one of the docker container IP addresses? Maybe you could share how your config is setup and what that IP corresponds to?

TFyre, 2024-09-05:
Looks like `PRINTER_ID` is not configured correctly. It should say `r2d2` instead of `00M09XXXXXXXXXX`

You can search the items below in your output:

```
 bambu-liveview-printer1-1 | Output #0, rtsp, to 'rtsp://mediamtx:8554/00M09XXXXXXXXXX':
```
```
no one is publishing to path 'r2d2'
```

sealltech, 2024-09-05:
Thank you for the quick response! 

So I changed the string that identifies what printer in the env.txt config to match the printers serial. Getting the same type of error with the expected new ID. Also verified that the printer ID in the config file is the same see below.

Error: closed: no one is publishing to path '00M09XXXXXXXXX'

________________________________________________________________________________________________

Snippit from console output:

bambu-liveview-printer1-1  | Input #0, rtsp, from 'rtsps://bblp:XXXXXXX@192.168.1.XX:322/streaming/live/1':
bambu-liveview-printer1-1  |   Metadata:
bambu-liveview-printer1-1  |     title           :
bambu-liveview-printer1-1  | rtsp stream server
bambu-liveview-printer1-1  |     comment         : Thu Sep  5 12:08:13 2024
bambu-liveview-printer1-1  |   Duration: N/A, start: 0.036000, bitrate: N/A
bambu-liveview-printer1-1  |   Stream #0:0: Video: h264 (Constrained Baseline), yuvj420p(pc, progressive), 1760x1080, 30 fps, 30 tbr, 90k tbn
bambu-liveview-printer1-1  |
bambu-liveview-printer1-1  | Stream mapping:
bambu-liveview-printer1-1  |   Stream #0:0 -> #0:0 (copy)
bambu-liveview-printer1-1  |
bambu-liveview-mediamtx-1  | 2024/09/05 17:19:29 INF [RTSP] [conn 172.18.0.5:37364] opened
bambu-liveview-mediamtx-1  | 2024/09/05 17:19:29 INF [RTSP] [session 80981d65] created by 172.18.0.5:37364
bambu-liveview-mediamtx-1  | 2024/09/05 17:19:29 INF [RTSP] [session 80981d65] is publishing to path '00M09XXXXXXXXX',
bambu-liveview-printer1-1  | Output #0, rtsp, to 'rtsp://mediamtx:8554/00M09XXXXXXXXX':

________________________________________________________________________________________________

env.txt new config
bambu.use-bouncy-castle=false
bambu.live-view-url=/_camerastream/

bambu.users.admin.password=XXXXXX
bambu.users.admin.role=admin

bambu.printers.00M09XXXXXXXXX.stream.live-view=true
bambu.printers.00M09XXXXXXXXX.enabled=true
bambu.printers.00M09XXXXXXXXX.name=Roboplop
bambu.printers.00M09XXXXXXXXX.device-id=00M09XXXXXXXXX
bambu.printers.00M09XXXXXXXXX.username=bblp
bambu.printers.00M09XXXXXXXXX.access-code=XXXXXXXXXX
bambu.printers.00M09XXXXXXXXX.ip=192.168.1.XX
bambu.printers.00M09XXXXXXXXX.use-ams=true
bambu.printers.00M09XXXXXXXXX.timelapse=true
bambu.printers.00M09XXXXXXXXX.bed-levelling=true
bambu.printers.00M09XXXXXXXXX.flow-calibration=true
bambu.printers.00M09XXXXXXXXX.vibration-calibration=true
bambu.printers.00M09XXXXXXXXX.model=x1c
bambu.printers.00M09XXXXXXXXX.mqtt.port=8883
bambu.printers.00M09XXXXXXXXX.mqtt.url=ssl://${bambu.printers.myprinter1.ip}:${bambu.printers.myprinter1.mqtt.port}
bambu.printers.00M09XXXXXXXXX.mqtt.report-topic=device/${bambu.printers.myprinter1.device-id}/report
bambu.printers.00M09XXXXXXXXX.mqtt.request-topic=device/${bambu.printers.myprinter1.device-id}/request
#Requesting full status interval
bambu.printers00M09XXXXXXXXX.mqtt.full-status=10m
bambu.printers.00M09XXXXXXXXX.ftp.port=990
bambu.printers.00M09XXXXXXXXX.ftp.url=ftps://${bambu.printers.myprinter1.ip}:${bambu.printers.myprinter1.ftp.port}
bambu.printers.00M09XXXXXXXXX.ftp.log-commands=false
bambu.printers.00M09XXXXXXXXX.stream.port=322
bambu.printers.00M09XXXXXXXXX.stream.url=ssl://${bambu.printers.myprinter1.ip}:${bambu.printers.myprinter1.stream.port}
#Restart stream if no images received interval
bambu.printers.00M09XXXXXXXXX.stream.watch-dog=5m

sealltech, 2024-09-05:
Cant say enough good things about this guy TFyre. Literally took time out of his day to troubleshoot the issue. For those with a similar issue here was the problem for me.
 
In the compose.yml variable PRINTER_ID: WHATEVERITIS make sure it matches the config in bambu-web-env.txt such as:

compose.yml snippit:
printer1:
        extends:
            file: common-liveview.yml
            service: liveview
        depends_on:
            - mediamtx
        environment:
            PRINTER_HOST: 192.168.1.XX
            PRINTER_ID: WHATEVERITIS
            PRINTER_ACCESS_CODE: XXXXXXXX

bambu-web-env.txt snippit:
bambu.printers.WHATEVERITIS.stream.live-view=true
bambu.printers.WHATEVERITIS.name=My Cool Printer Name
bambu.printers.WHATEVERITIS.device-id=XXXXXXXXXXXXXX
bambu.printers.WHATEVERITIS.access-code=XXXXXXXX
bambu.printers.WHATEVERITIS.ip=192.168.1.XX
bambu.printers.WHATEVERITIS.model=x1c

Thanks again TFyre its all now working!

closed 2024-09-05 by sealltech