TFyre/bambu-farm #153: Print

csiguszka, 2024-09-08:
Hi!

I need some help with a project I'm working on. I have several Bambu Lab printers, and I'm developing software to automate the printing process. My software generates STL files, slices them to create G-code, and then uploads these G-code files to the printers via FTP. However, I’m currently unable to start the print jobs automatically; I have to do it manually from the printer's panel.

Can you help me figure out how to programmatically start the print job? How did you do exactly?

TFyre, 2024-09-11:
You need to send a `MQTT` message that looks something like this: https://github.com/TFyre/bambu-farm/blob/eb22074e2fdbd99df3093952ac475ab380734ef3/bambu/src/main/java/com/tfyre/bambu/printer/BambuPrinterImpl.java#L472

csiguszka, 2024-09-11:
I did it! Thank you so much!

closed 2024-09-11 by csiguszka