TFyre/bambu-farm #154: Feature Request - "print again" function?

TheBasedDoge, 2024-09-15:
I love this software, it makes managing my small farm a breeze. Two feature requests that would make it even better.

- Add a "print again" button on the main control page, so the same gcode can easily be run again without navigating back to the  SD card.
- Memory for the checkboxes before starting a print. For example, I have to uncheck timelapse every time I start a print file.

TFyre, 2025-04-10:
* For "print again" please check #254 feature when printing from the batch page
* For memory of checkboxes, please README configuration

[Printer Section](https://github.com/TFyre/bambu-farm?tab=readme-ov-file#printer-section)
```properties
bambu.printers.myprinter1.timelapse=false
```

[Batch print](https://github.com/TFyre/bambu-farm?tab=readme-ov-file#batch-print-section)
```properties
bambu.batch-print.timelapse=false
```

closed 2025-04-10 by TFyre

TheBasedDoge, 2025-04-10:
Would it be possible to add the print again button on the main dashboard screen? For example, sometimes I have the printers running different jobs so in this case batch print wouldn't make sense, but I still need each printer to redo the same job multiple times.