TFyre/bambu-farm #158: Minor Request: Can the progress bar be relocatable on the dashboard as a setting?

VIDGuide, 2024-09-18:
2 of my printers have multi-AMS's and this pushes the progress/time-remaining box off the bottom of the screen for my dashboard. 
I'm less interested in the AMS status (though I do like having it there, I'd prefer not to have to turn it off in the .env), but I am very interested to see the progress bar at a glance. 

Would be nice if there was a config option that would put it up near the control buttons, or above the stream perhaps.

VIDGuide, 2024-09-20:
#159 Added PR for a patch to handle this, if it's suitable to your style/intentions. 
Added an .env var to control position, "above-image", "below-image", "bottom" (default, current location)

TFyre, 2025-01-04:
Closing this as the order can be configured using CSS [(readme.md)](https://github.com/TFyre/bambu-farm?tab=readme-ov-file#ordering-items-inside-printer-box):

```css
.dashboard-printer .image {
    order: 3;
}
.dashboard-printer .status {
    order: 4;
}
.dashboard-printer .filaments {
    order: 1;
}
```

closed 2025-01-04 by TFyre