TFyre/bambu-farm #163: Missing PetG-HF

glennswest, 2024-09-23:
Looks like the available filaments need to be updated.
Im using relatively new Bambu PetG Basic, and it shows as unknown in bambu-farm with latest release.
Shows as PetG in bambu app.
Im also getting a printer error, when status's look clean, and last print finished normally. (Could just be the unknown filament)

[image]

glennswest, 2024-09-23:
[image]

glennswest, 2024-09-26:
Will work on doing a PR for additional Filament types.

TFyre, 2025-01-04:
Please share the details from logs in the UI:

I need the highlighted section to configure the filament codes in [Filament.java](https://github.com/TFyre/bambu-farm/blob/main/bambu/src/main/java/com/tfyre/bambu/printer/Filament.java) 

[image]

TFyre, 2025-01-04:
Closing the issue for now, please re-open with relevant information

closed 2025-01-04 by TFyre