TFyre/bambu-farm #180: Help for installation

ntjaz, 2024-11-19:
Dear all,

I would really appreciate your help with the installation. I´m a complete noob with linux, shell, docker and the other crazy things.
So please be kind with me.

I set up a raspberry pi with docker.
Run Putty and install java as shown in description.
Download bambu-web-*-runner.jar

But then the struggle startet, how to create .env file?
How to change and save it?

As i said, first time ever trying something like this.
Maybe someone could make a step by step guide? Or a short screencast? 

Thx for you support!

TFyre, 2024-12-01:
Using putty you can use any console based editor (`vim` / `nano`) to edit the .env file

You can also edit/create the file on windows and scp (secure copy - `pscp` from putty) it to the raspberry pi.

ntjaz, 2024-12-02:
Hey Francois,

thx for your help.
i managed to create .env file via nano.

now when i try to run with "java -jar bambu-web-1.3.0-runner.jar"
i get one ERROR which is this one

2024-12-02 09:19:40,775 ERROR [com.tfy.bam.ssl.NoopTrustSocketFactory] (main) Using INSECURE com.tfyre.bambu.ssl.NoopTrustSocketFactory

do you have any idear what this could be?

big thx for your support!

TFyre, 2024-12-02:
That is an error log, but does not indicate a problem ;)

what seems to be the issue? Please share a "stack trace" of sorts

ntjaz, 2024-12-02:
hey,

here is the trace. thx

java -jar bambu-web-1.3.0-runner.jar
__  ____  __  _____   ___  __ ____  ______
 --/ __ \/ / / / _ | / _ \/ //_/ / / / __/
 -/ /_/ / /_/ / __ |/ , _/ ,< / /_/ /\ \
--\___\_\____/_/ |_/_/|_/_/|_|\____/___/
2024-12-02 09:19:38,185 INFO  [io.und.servlet] (main) Initializing AtmosphereFramework
2024-12-02 09:19:38,242 INFO  [com.vaa.flo.ser.sta.ServletDeployer] (main) Skipping automatic servlet registration because there is already a Vaadin servlet with the name com.vaadin.quarkus.QuarkusVaadinServlet
2024-12-02 09:19:38,302 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereHandler com.vaadin.flow.server.communication.PushAtmosphereHandler mapped to context-path: /*
2024-12-02 09:19:38,302 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed the following AtmosphereInterceptor mapped to AtmosphereHandler com.vaadin.flow.server.communication.PushAtmosphereHandler
2024-12-02 09:19:38,458 INFO  [org.atm.cpr.AtmosphereFramework] (main) Atmosphere is using org.atmosphere.util.VoidAnnotationProcessor for processing annotation
2024-12-02 09:19:38,488 INFO  [org.atm.uti.ForkJoinPool] (main) Using ForkJoinPool  java.util.concurrent.ForkJoinPool. Set the org.atmosphere.cpr.broadcaster.maxAsyncWriteThreads to -1 to fully use its power.
2024-12-02 09:19:38,516 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed WebSocketProtocol org.atmosphere.websocket.protocol.SimpleHttpProtocol
2024-12-02 09:19:38,532 INFO  [org.atm.con.JSR356AsyncSupport] (main) JSR 356 Mapping path /VAADIN/push
2024-12-02 09:19:38,544 INFO  [io.und.websockets] (main) UT026005: Adding programmatic server endpoint class org.atmosphere.container.JSR356Endpoint for path /VAADIN/push
2024-12-02 09:19:38,550 INFO  [io.und.websockets] (main) UT026005: Adding programmatic server endpoint class org.atmosphere.container.JSR356Endpoint for path /VAADIN/push/
2024-12-02 09:19:38,550 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installing Default AtmosphereInterceptors
2024-12-02 09:19:38,552 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.CorsInterceptor : CORS Interceptor Support
2024-12-02 09:19:38,554 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.CacheHeadersInterceptor : Default Response's Headers Interceptor
2024-12-02 09:19:38,557 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.PaddingAtmosphereInterceptor : Browser Padding Interceptor Support
2024-12-02 09:19:38,559 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.AndroidAtmosphereInterceptor : Android Interceptor Support
2024-12-02 09:19:38,562 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.HeartbeatInterceptor : Heartbeat Interceptor Support
2024-12-02 09:19:38,564 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.SSEAtmosphereInterceptor : SSE Interceptor Support
2024-12-02 09:19:38,566 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.JSONPAtmosphereInterceptor : JSONP Interceptor Support
2024-12-02 09:19:38,569 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.JavaScriptProtocol : Atmosphere JavaScript Protocol
2024-12-02 09:19:38,570 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.WebSocketMessageSuspendInterceptor : org.atmosphere.interceptor.WebSocketMessageSuspendInterceptor
2024-12-02 09:19:38,571 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.OnDisconnectInterceptor : Browser disconnection detection
2024-12-02 09:19:38,573 INFO  [org.atm.cpr.AtmosphereFramework] (main)  org.atmosphere.interceptor.IdleResourceInterceptor : org.atmosphere.interceptor.IdleResourceInterceptor
2024-12-02 09:19:38,573 INFO  [org.atm.cpr.AtmosphereFramework] (main) Set org.atmosphere.cpr.AtmosphereInterceptor.disableDefaults to disable them.
2024-12-02 09:19:38,574 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor CORS Interceptor Support with priority FIRST_BEFORE_DEFAULT
2024-12-02 09:19:38,575 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor Default Response's Headers Interceptor with priority AFTER_DEFAULT
2024-12-02 09:19:38,576 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor Browser Padding Interceptor Support with priority AFTER_DEFAULT
2024-12-02 09:19:38,576 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor Android Interceptor Support with priority AFTER_DEFAULT
2024-12-02 09:19:38,576 INFO  [org.atm.int.HeartbeatInterceptor] (main) HeartbeatInterceptor configured with padding value 'X', client frequency 0 seconds and server frequency 60 seconds
2024-12-02 09:19:38,577 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor Heartbeat Interceptor Support with priority AFTER_DEFAULT
2024-12-02 09:19:38,577 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor SSE Interceptor Support with priority AFTER_DEFAULT
2024-12-02 09:19:38,577 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor JSONP Interceptor Support with priority AFTER_DEFAULT
2024-12-02 09:19:38,577 INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereInterceptor Atmosphere JavaScript Protocol with priority AFTER_DEFAULT
[... cut ...]
2024-12-02 09:21:27,608 INFO  [com.tfy.bam.pri.BambuPrintersImpl] (Shutdown thread) Stopping Printers
2024-12-02 09:21:27,610 INFO  [com.tfy.bam.pri.BambuPrintersImpl] (Shutdown thread) myprinter1: stopping
2024-12-02 09:21:27,610 INFO  [com.tfy.bam.pri.BambuPrinterStream] (Shutdown thread) myprinter1: stopping
2024-12-02 09:21:27,612 INFO  [com.tfy.bam.pri.BambuPrintersImpl] (Shutdown thread) myprinter1: stopped
2024-12-02 09:21:27,665 INFO  [io.quarkus] (Shutdown thread) bambu-web stopped in 0.228s

TFyre, 2024-12-02:
nothing wrong with the stacktrace, was running fine until someone pressed CTRL+C to stop it:
```
^C2024-12-02 09:21:27,495 INFO [org.apa.cam.imp.eng.AbstractCamelContext] (Shutdown thread) Apache Camel 4.8.0 (camel-1) is shutting down (timeout:45s)
```

ntjaz, 2024-12-02:
Hi Francois,

i got it running. really appreciate your help. thx a lot.

i have only one more issue, when i create a printer in .env file
i use .3DP-P1S-03. instead of .printer1.

when i runt the web interface, it shows 2 printers 
one with big letters like i defined in .env
one with small letters

do you have any idear why it does it?
thx

TFyre, 2025-01-04:
Fixed in commit: 175ad98

closed 2025-01-04 by TFyre