TFyre/bambu-farm #183: Cloud mode is not working - error 403

dfjaapie, 2024-11-27:
Recently I upgraded my printers and moved to using clode mode but unfortunatly this doesn't work. 

I have tried using option 1 with the login information and option 2 with the token. In the docker log i'm seeing this error

2024-11-27 08:47:48 Caused by: java.lang.RuntimeException: java.lang.RuntimeException: org.jboss.resteasy.reactive.ClientWebApplicationException: Received: 'Forbidden, status code 403' when invoking REST Client method: 'com.tfyre.bambu.CloudApi#login'

What could this be?

Nikzda, 2024-11-30:
I get the same Error. It all worked perfectly for some weeks

Lim0H, 2024-12-08:
> I get the same Error. It all worked perfectly for some weeks
> 
> Edit: Switching from user credentials to token fixed it for me

Hello. How you get this token. I tryied bambulab.com and makerworld for getting token from cookies, but this don't work for me. JWT decode don't work for me tokens from this sites

TFyre, 2025-01-07:
https://github.com/TFyre/bambu-farm/releases/tag/v1.5.0

Please grab and test version 1.5.0 from the link above and let me know if it works now.

Have a read through the [Cloud Section](https://github.com/TFyre/bambu-farm#cloud-section) to obtain the new configuration options:
```properties
bambu.cloud.enabled=true
bambu.cloud.username=u_12345
bambu.cloud.token=AA...
```

PietroFurlan, 2025-01-08:
It works!
Thank you.

closed 2025-04-10 by TFyre