TFyre/bambu-farm #184: Docker container

EdwardChamberlain, 2024-11-29:
Would it be possible to have this built as a docker container for easier deployment?

Locxion, 2024-11-29:
There is already a docker solution
https://github.com/TFyre/bambu-farm/blob/main/docker/bambu-liveview/README.md

EdwardChamberlain, 2024-11-29:
@Locxion Is there a reason this cant be deployed as a single container? Ideally id like to run this under unraid so using docker compose is a pain.

Locxion, 2024-11-29:
i have no idea
the livecam containers work per printer
i dont know why the other services are seperated

TFyre, 2024-12-01:
> @Locxion Is there a reason this cant be deployed as a single container? Ideally id like to run this under unraid so using docker compose is a pain.

No reason, but it defeats the purpose of having a docker compose setup based on thin layers with good isolation.

Feel free to submit a PR to add single-container-functionality 😄

TFyre, 2025-01-04:
Closing stale issue, please re-open if required

closed 2025-01-04 by TFyre