TFyre/bambu-farm #197: JWT error , chrome cookie get

Glianze, 2024-12-12:
org.jose4j.jwt.consumer.InvalidJwtException: JWT processing failed. Additional details: [[17] Unable to process JOSE object (cause: org.jose4j.lang.JoseException: Invalid JOSE Compact Serialization. Expecting either 3 or 5 parts for JWS or JWE respectively but was 1.): AAA7X3kiP67axOzUxLziXj06KpPip-lYwlMmx-xd_6nQJFCbZRQUcgG2Pu40bpzHYMsg_3tYGedIfQyOXUOstFfzf72JFJa9Zmiv-v7W-pBroZ0HjxtKIoSdEGJ9V5wFSTbrXZ2hX2Xx2Zgk]

How can I solve the error problem that prevents us from remotely obtaining printer information?

[image]
[image]

TFyre, 2024-12-12:
That JWT token doesnt look valid, you can test it here: https://jwt.io/

TFyre, 2025-01-04:
Closing stale issue, please re-open if required