TFyre/bambu-farm #198: Failed to start application - A job with this identity is already scheduled

TomKamin1, 2024-12-12:
Hi. I did a fresh install (I had it working before but my server needed updatign and it broke) and im getting this error and am unable to fix it. I will be grateful for any help.

2024-12-12 12:25:50,599 ERROR [org.apa.cam.qua.mai.CamelMainRuntime] (main) Failed to start application: java.lang.IllegalStateException: A job with this identity is already scheduled: com.tfyre.bambu.printer.BambuPrinterImpl.requestFullStatus#Alpha A1
	at io.quarkus.scheduler.runtime.SimpleScheduler.newJob(SimpleScheduler.java:208)
	at com.tfyre.bambu.printer.BambuPrinterImpl.setup(BambuPrinterImpl.java:126)
	at com.tfyre.bambu.printer.BambuPrintersImpl.newPrinter(BambuPrintersImpl.java:73)
	at com.tfyre.bambu.printer.BambuPrintersImpl_ClientProxy.newPrinter(Unknown Source)
	at com.tfyre.bambu.camel.CamelController.configurePrinter(CamelController.java:89)
	at java.base/java.util.HashMap.forEach(HashMap.java:1429)
	at com.tfyre.bambu.camel.CamelController.configure(CamelController.java:62)
	at org.apache.camel.builder.RouteBuilder.checkInitialized(RouteBuilder.java:809)
	at org.apache.camel.builder.RouteBuilder.configureRoutes(RouteBuilder.java:733)
	at org.apache.camel.builder.RouteBuilder.prepareModel(RouteBuilder.java:649)
	at org.apache.camel.builder.RouteBuilder.addRoutesToCamelContext(RouteBuilder.java:667)
	at com.tfyre.bambu.camel.CamelController_ClientProxy.addRoutesToCamelContext(Unknown Source)
	at org.apache.camel.impl.engine.AbstractCamelContext.addRoutes(AbstractCamelContext.java:995)
	at org.apache.camel.main.RoutesConfigurer.addDiscoveredRoutes(RoutesConfigurer.java:291)
	at org.apache.camel.main.RoutesConfigurer.configureRoutes(RoutesConfigurer.java:265)
	at org.apache.camel.quarkus.main.CamelMain.configureRoutes(CamelMain.java:67)
	at org.apache.camel.main.BaseMainSupport.postProcessCamelContext(BaseMainSupport.java:782)
	at org.apache.camel.quarkus.main.CamelMain.initCamelContext(CamelMain.java:113)
	at org.apache.camel.quarkus.main.CamelMain.doInit(CamelMain.java:83)
	at org.apache.camel.support.service.BaseService.init(BaseService.java:85)
	at org.apache.camel.quarkus.main.CamelMain.startEngine(CamelMain.java:134)
	at org.apache.camel.quarkus.main.CamelMainRuntime.start(CamelMainRuntime.java:49)
	at org.apache.camel.quarkus.core.CamelBootstrapRecorder.start(CamelBootstrapRecorder.java:45)
	at io.quarkus.deployment.steps.CamelBootstrapProcessor$boot173480958.deploy_0(Unknown Source)
	at io.quarkus.deployment.steps.CamelBootstrapProcessor$boot173480958.deploy(Unknown Source)
	at io.quarkus.runner.ApplicationImpl.doStart(Unknown Source)
	at io.quarkus.runtime.Application.start(Application.java:101)
	at io.quarkus.runtime.ApplicationLifecycleManager.run(ApplicationLifecycleManager.java:119)
	at io.quarkus.runtime.Quarkus.run(Quarkus.java:71)
	at io.quarkus.runtime.Quarkus.run(Quarkus.java:44)
	at io.quarkus.runtime.Quarkus.run(Quarkus.java:124)
	at io.quarkus.runner.GeneratedMain.main(Unknown Source)

2024-12-12 12:25:50,605 INFO  [com.tfy.bam.pri.BambuPrintersImpl] (main) Stopping Printers
2024-12-12 12:25:50,619 ERROR [io.qua.run.Application] (main) Failed to start application: java.lang.RuntimeException: Failed to start quarkus
	at io.quarkus.runner.ApplicationImpl.doStart(Unknown Source)
	at io.quarkus.runtime.Application.start(Application.java:101)
	at io.quarkus.runtime.ApplicationLifecycleManager.run(ApplicationLifecycleManager.java:119)
	at io.quarkus.runtime.Quarkus.run(Quarkus.java:71)
	at io.quarkus.runtime.Quarkus.run(Quarkus.java:44)
	at io.quarkus.runtime.Quarkus.run(Quarkus.java:124)
	at io.quarkus.runner.GeneratedMain.main(Unknown Source)
Caused by: java.lang.RuntimeException: java.lang.RuntimeException: java.lang.IllegalStateException: A job with this identity is already scheduled: com.tfyre.bambu.printer.BambuPrinterImpl.requestFullStatus#Alpha A1
	at org.apache.camel.quarkus.core.CamelBootstrapRecorder.start(CamelBootstrapRecorder.java:47)
	at io.quarkus.deployment.steps.CamelBootstrapProcessor$boot173480958.deploy_0(Unknown Source)
	at io.quarkus.deployment.steps.CamelBootstrapProcessor$boot173480958.deploy(Unknown Source)
	... 7 more
Caused by: java.lang.RuntimeException: java.lang.IllegalStateException: A job with this identity is already scheduled: com.tfyre.bambu.printer.BambuPrinterImpl.requestFullStatus#Alpha A1
	at org.apache.camel.quarkus.main.CamelMainRuntime.start(CamelMainRuntime.java:65)
	at org.apache.camel.quarkus.core.CamelBootstrapRecorder.start(CamelBootstrapRecorder.java:45)
	... 9 more
Caused by: java.lang.IllegalStateException: A job with this identity is already scheduled: com.tfyre.bambu.printer.BambuPrinterImpl.requestFullStatus#Alpha A1
	at io.quarkus.scheduler.runtime.SimpleScheduler.newJob(SimpleScheduler.java:208)
	at com.tfyre.bambu.printer.BambuPrinterImpl.setup(BambuPrinterImpl.java:126)
	at com.tfyre.bambu.printer.BambuPrintersImpl.newPrinter(BambuPrintersImpl.java:73)
	at com.tfyre.bambu.printer.BambuPrintersImpl_ClientProxy.newPrinter(Unknown Source)
	at com.tfyre.bambu.camel.CamelController.configurePrinter(CamelController.java:89)
	at java.base/java.util.HashMap.forEach(HashMap.java:1429)
	at com.tfyre.bambu.camel.CamelController.configure(CamelController.java:62)
	at org.apache.camel.builder.RouteBuilder.checkInitialized(RouteBuilder.java:809)
	at org.apache.camel.builder.RouteBuilder.configureRoutes(RouteBuilder.java:733)
	at org.apache.camel.builder.RouteBuilder.prepareModel(RouteBuilder.java:649)
	at org.apache.camel.builder.RouteBuilder.addRoutesToCamelContext(RouteBuilder.java:667)
	at com.tfyre.bambu.camel.CamelController_ClientProxy.addRoutesToCamelContext(Unknown Source)
	at org.apache.camel.impl.engine.AbstractCamelContext.addRoutes(AbstractCamelContext.java:995)
	at org.apache.camel.main.RoutesConfigurer.addDiscoveredRoutes(RoutesConfigurer.java:291)
	at org.apache.camel.main.RoutesConfigurer.configureRoutes(RoutesConfigurer.java:265)
	at org.apache.camel.quarkus.main.CamelMain.configureRoutes(CamelMain.java:67)
	at org.apache.camel.main.BaseMainSupport.postProcessCamelContext(BaseMainSupport.java:782)
	at org.apache.camel.quarkus.main.CamelMain.initCamelContext(CamelMain.java:113)
	at org.apache.camel.quarkus.main.CamelMain.doInit(CamelMain.java:83)
	at org.apache.camel.support.service.BaseService.init(BaseService.java:85)
	at org.apache.camel.quarkus.main.CamelMain.startEngine(CamelMain.java:134)
	at org.apache.camel.quarkus.main.CamelMainRuntime.start(CamelMainRuntime.java:49)
	... 10 more

TFyre, 2024-12-12:
Looks like you have duplicate / invalid printer config.

Please refer to this section: https://github.com/TFyre/bambu-farm?tab=readme-ov-file#printer-section

TomKamin1, 2024-12-18:
Hello? Any ideas? I just cannot get it working :/ 

> Looks like you have duplicate / invalid printer config.
> 
> Please refer to this section: https://github.com/TFyre/bambu-farm?tab=readme-ov-file#printer-section

TFyre, 2024-12-18:
#192 

Please check this issue. I believe you have a duplicate entey in the config somewhere

TomKamin1, 2024-12-18:
Thank you, it was the uppercase characters, for some reason that worked before, but now it would cause the errors. Thank you again 

> #192
> 
> Please check this issue. I believe you have a duplicate entey in the config somewhere

TFyre, 2024-12-18:
Thx for the feedback, will update the documentation

closed 2025-01-04 by TFyre