TFyre/bambu-farm #203: The webview works but there is no video...

jsc255, 2024-12-26:
I installed everything correctly and I think I got all the config files correct. When I login to the dashboard I can see my 4 X1C printer stats but no video. Yes installed the Docker Compose environment for X1C printers and set the configs. I have tried turning on the "LAN only" and "LAN Mode Liveview" then rebooting the X1c's. Can I just turn on "LAN Mode Liveview" and still have this work? What am I doing wrong?

System info:
Ubuntu 24

Config below:
.env 
`quarkus.http.host=0.0.0.0
quarkus.http.port=8080
bambu.printers.myprinter1.device-id=*********HIDDEN**********
bambu.printers.myprinter1.access-code=74ef8d53
bambu.printers.myprinter1.ip=192.168.1.33
bambu.printers.myprinter1.stream.live-view=true
bambu.printers.myprinter2.device-id=*********HIDDEN**********
bambu.printers.myprinter2.access-code=ba94ce20
bambu.printers.myprinter2.ip=192.168.1.96
bambu.printers.myprinter2.stream.live-view=true
bambu.printers.myprinter3.device-id=*********HIDDEN**********
bambu.printers.myprinter3.access-code=e9417977
bambu.printers.myprinter3.ip=192.168.1.56
bambu.printers.myprinter3.stream.live-view=true
bambu.printers.myprinter4.device-id=*********HIDDEN**********
bambu.printers.myprinter4.access-code=65e01082
bambu.printers.myprinter4.ip=192.168.1.94
bambu.printers.myprinter4.stream.live-view=true
bambu.users.admin.password=admin
bambu.users.admin.role=admin
bambu.remote-view=true
bambu.dashboard.remote-view=true
`

TFyre, 2025-01-04:
Please share the following:
* docker compose file
* detailed error logs

TFyre, 2025-04-10:
Closing stale issue, please re-open if required

closed 2025-04-10 by TFyre