TFyre/bambu-farm #213: Can't Login with USER:admin PW:admin

Marco-Polo20, 2025-01-18:
Good day everyone,

I am facing the issue that I cannot log into the interface (http://127.0.0.1:8080) using the default username and password.
When starting, the CMD outputs 

_"Routes startup (started:0)"._

And after one or more trys it says

_"ERROR [com.tfy.bam.sec.SecurityUtils] (executor-thread-1) Login failed for [admin] - io.quarkus.security.AuthenticationFailedException"._

The .env file is located in the same folder as the .jar file, and it was created based on the "minimal config". The IP, ID, and Key are working in a Python API.

JDK version: 21.0.5
The setup was tested on Windows 10, 11, and Linux (Debian).

Unfortunately, I couldn't find anything online regarding my issue.
Where could I have made a mistake?

Thank you very much for your help.

TFyre, 2025-01-18:
Can you please share your .env file (mask sensitive info)

User: admin
pass: admin

Tested & worked with:
```properties
bambu.users.admin.password=admin
bambu.users.admin.role=admin
```

Marco-Polo20, 2025-01-18:
Thank you for your message.

It has been resolved.
The .env file should not have a name. It should simply be ".env", not "NAME.env" or "config.env".

Sorry for the disturbance. 👍

closed 2025-01-18 by TFyre