TFyre/bambu-farm #216: Unable to get camera working in docker

spuder, 2025-01-22:
First of all, thank you so much for making this software. I've tried HomeAssistant, OctoEverywhere but they just don't give me quite what I'm looking for. This software looks perfect!

I'm finding that I am unable to stream video from my a1 mini in docker compose. 

**Is LAN LiveView an X1 only feature?** I've searched though the menus and I'm unable to find this setting on an A1 Mini. I suspect it might be an X1 only feature. Please correct me if I'm wrong. Based on the fact that other software is able to stream the webcam feed I suspect the problem is with my configuration, not the printer. 

<details>

<summary>mediamtx.yaml</summary>

All settings are defaults except

```ruby
webrtc: yes
webrtcAddress: :8889
webrtcEncryption: no
webrtcServerKey: server.key
webrtcServerCert: server.crt
webrtcAllowOrigin: '*'
webrtcTrustedProxies: []
webrtcLocalUDPAddress: #:8189
webrtcLocalTCPAddress: :8189
webrtcAdditionalHosts: [127.0.0.1, 10.89.0.75, 10.89.0.85, 192.168.81.146, bambu.redacted.com]
```

</details>

<details>

<summary>common-liveview.yaml</summary>

config

```ruby
version: '3'

services:
    liveview:
        image: linuxserver/ffmpeg
        restart: always
        environment:
            PRINTER_HOST: DONT_TOUCH
            PRINTER_USER: bblp
            PRINTER_ID: DONT_TOUCH
            PRINTER_ACCESS_CODE: DONT_TOUCH
            PRINTER_PORT: 322
            PRINTER_URL: /streaming/live/1
            RTSP_SERVER: mediamtx:8554
        entrypoint:
            - "bash"
            - "-c"
            - >
              /usr/local/bin/ffmpeg
              -timeout 30000000
              -re -stream_loop -1
              -i rtsps://$${PRINTER_USER}:$${PRINTER_ACCESS_CODE}@$${PRINTER_HOST}:$${PRINTER_PORT}$${PRINTER_URL}
              -c:v copy -f rtsp -rtsp_transport tcp
              rtsp://$${RTSP_SERVER}/$${PRINTER_ID}
```

</details>

<details>

<summary>mediamtx</summary>

```ruby
version: '3.8'

services:
    mediamtx:
        image: bluenviron/mediamtx:1.8.2
        restart: always
        volumes:
            - ./mediamtx.yaml:/mediamtx.yaml
        ports:
            #RTSP
            #- '8554:8554' 

            #WebRTC
            - '8189:8189/tcp'
            - '8189:8189/udp'
            #- '8888:8888'
            - '8889:8889'

    bambuweb:
        image: azul/zulu-openjdk:21-latest
        restart: always
        volumes:
            - ./bambu-web-1.5.0-runner.jar:/bambu-web-runner.jar
            - ./bambu-web-env.txt:/.env
        ports:
            - '8081:8080'
        entrypoint: /usr/bin/java -Djdk.tls.useExtendedMasterSecret=false -jar bambu-web-runner.jar

    nginx:
        image: nginx:1.26-alpine
        restart: always
        volumes:
            - ./reverse-proxy.conf:/etc/nginx/conf.d/default.conf
        ports:
            - '8080:80'
        depends_on:
            - mediamtx
            - bambuweb

    printer1:
        extends:
            file: common-liveview.yaml
            service: liveview
        depends_on:
            - mediamtx
        environment:
            PRINTER_HOST: ${PRINTER1_HOST}
            PRINTER_ID: ${PRINTER1_ID}
            PRINTER_ACCESS_CODE: ${PRINTER1_ACCESS_CODE}

```

</details>

<details>

<summary>reverse-proxy.conf</summary>

All settings are defaults except

```ruby
map $http_upgrade $connection_upgrade {
        default upgrade;
        #'' close;
	'' '';
}

upstream mediamtx {
	keepalive 32;
        server mediamtx:8889;
}

upstream bambuweb {
	keepalive 32;
        server bambuweb:8080;
}

server {
    listen       80;
    listen  [::]:80;
    server_name  reverse-proxy;
    access_log off;
    root   /usr/share/nginx/html;
    index  index.html index.htm;

    location /_camerastream/ {
        rewrite /_camerastream/(.*) /$1  break;
        proxy_redirect / /_camerastream/;
        absolute_redirect off;
        proxy_pass http://mediamtx/;
        proxy_http_version 1.1;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection $connection_upgrade;
        proxy_set_header Host $host;
        # for websockets
        proxy_read_timeout 5m;
    }

    location / {
        client_max_body_size 20m;
        proxy_pass http://bambuweb;
        proxy_http_version 1.1;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection $connection_upgrade;
        proxy_set_header Host $host;
        # for websockets
        proxy_read_timeout 5m;
    }

    error_page 404 /404.html;
        location = /40x.html {
    }

    error_page   500 502 503 504  /50x.html;
    location = /50x.html {
        root   /usr/share/nginx/html;
    }

}

```

</details>

The key config is stored as environment variables in a .envrc file that is loaded before `docker compse up`

```
export PRINTER1_HOST=192.168.81.108
export PRINTER1_ID=a1mini-1
export PRINTER1_ACCESS_CODE=xxxxxx
```

There is also a bambu-web-env.txt

```
bambu.live-view-url=/_camerastream/
bambu.printers.a1mini-1.stream.live-view=true

bambu.dark-mode=true

quarkus.http.host=0.0.0.0
quarkus.http.port=8080

bambu.printers.a1mini-1.name=a1-mini-1

bambu.printers.a1mini-1.device-id=xxxxxxxxxxxxxx
bambu.printers.a1mini-1.access-code=xxxxxxx
bambu.printers.a1mini-1.ip=192.168.81.108

bambu.users.admin.password=xxxxxx
bambu.users.admin.role=xxxx

```

The key error is `no one is publishing to path 'a1mini-1'`

```
mediamtx-1  | 2025/01/22 17:23:28 INF [WebRTC] [session 56bb46cd] closed: no one is publishing to path 'a1mini-1'
bambuweb-1  | 2025-01-22 17:23:29,719 INFO  [org.apa.cam.imp.eng.AbstractCamelContext] (executor-thread-1) Started consumer-a1-mini-1 (paho://device/0309CA451301801/report)
bambuweb-1  | 2025-01-22 17:23:30,876 INFO  [com.tfy.bam.pri.BambuPrintersImpl] (executor-thread-1) a1-mini-1: started
mediamtx-1  | 2025/01/22 17:23:31 INF [WebRTC] [session 4956b98f] created by 10.89.0.95:47004
mediamtx-1  | 2025/01/22 17:23:31 INF [WebRTC] [session 4956b98f] closed: no one is publishing to path 'a1mini-1'
```

I'll happily add documentation to help others as it seems I'm not the first one running into this problem. 

Do you have any ideas what I am missing?

TFyre, 2025-01-22:
Yes LanView is an X only thing (proper camera video stream).... a/p/s models uses mjpeg.... basically custom auth and then a stream of sequential jpegs...

I havent had a need for it, but it would be possible to build 'middleware' that grabs the mjpeg stream and make it available via mediamtx using ffmpeg

spuder, 2025-01-22:
Thanks for the response. I can add some documentation clarifying that only X1 users need to enable LanView. 

So am I understanding that it's not possible to get the video (really just images) from an A1/P1 printer?

closed 2025-04-10 by TFyre