TFyre/bambu-farm #217: No printers Cloud

thomveen, 2025-01-23:
I have gone through all the steps carefully. I don't get an error message but I still don't see my printers with the cloud.

TFyre, 2025-01-23:
Please share your .env file after masking sensitive information

thomveen, 2025-01-24:
quarkus.http.host=0.0.0.0
quarkus.http.port=8080

bambu.cloud.enabled=true
bambu.cloud.username=u_27user
bambu.cloud.token=AAtoken

#bambu.printers.myprinter1.device-id=REPLACE_WITH_DEVICE_SERIAL
#bambu.printers.myprinter1.access-code=REPLACE_WITH_DEVICE_ACCESSCODE
#bambu.printers.myprinter1.ip=REPLACE_WITH_DEVICE_IP

bambu.users.admin.password=admin
bambu.users.admin.role=admin

bambu.batch-print.timelapse=true
bambu.batch-print.bed-levelling=true
bambu.batch-print.flow-calibration=true
bambu.batch-print.vibration-calibration=true

bambu.preheat[0].name=Off 0/0
bambu.preheat[0].bed=0
bambu.preheat[0].nozzle=0
bambu.preheat[1].name=PLA 55/220
bambu.preheat[1].bed=55
bambu.preheat[1].nozzle=220
bambu.preheat[2].name=ABS 90/270
bambu.preheat[2].bed=90
bambu.preheat[2].nozzle=270

# defaults to true, when false, disables remote view globally
bambu.remote-view=true

# defaults to true, when false, disables remote view for dashboard, but will still be available in detail view
bambu.dashboard.remote-view=true

# defaults to true, when false, disables per printer
#bambu.printers.myprinter1.stream.enable=true

TFyre, 2025-01-24:
You need to configure a printer: uncomment and complete the following section:

```properties
#bambu.printers.myprinter1.device-id=REPLACE_WITH_DEVICE_SERIAL
#bambu.printers.myprinter1.access-code=REPLACE_WITH_DEVICE_ACCESSCODE
#bambu.printers.myprinter1.ip=REPLACE_WITH_DEVICE_IP
```

thomveen, 2025-01-27:
Thank you for your feedback I now indeed have the printers in it. but what exactly is the advantage of bambu cloud? I expected him to automatically add my printers to the configuration? I also have problems with Batch Print. I can upload and send a file to the printer but the printer will not print.

closed 2025-04-10 by TFyre