TFyre/bambu-farm #219: Add Support for Custom Third-party Filaments

FuturePrints, 2025-01-27:
Enable the platform to read filament types directly from the machine. Currently, I have to set my profiles on the AMS to "Generic" to use batch printing with custom third-party filaments.

**Problem:**
The current setup results in the following error:
[image]

[image]

**Handy App Active Configuration:**

[image]

**Proposed Solution:**

- Enhance the platform to detect and read the filament type automatically from the machine.

- Ensure compatibility with third-party filaments by enabling batch printing without requiring manual profile adjustments.

This improvement would significantly streamline the process and enhance usability for users who work with a variety of filament brands.

TFyre, 2025-01-27:
Please have a look at #163 

Filaments are being detected automatically, but they should be known before hand 😄

FuturePrints, 2025-01-27:
But it is known by the printer not by IDX but using filament type

Maybe add a check for IDX and filament type?

[image]

[image]

mattwarr, 2026-09-08:
some way to add in new filaments, be they 3rd party or even new BambuLabs filaments would be massively useful . I find it quite frustrating that BBL PLA silk+ shows up as unknown

mattwarr, 2026-09-23:
ok, so i made it work in identifying PLA silk+ so that has improved my QOL :)

TFyre, 2026-09-24:
Can you elaborate on how you made it work?

mattwarr, 2026-09-24:
> Can you elaborate on how you made it work?

I added 
BAMBU_PLA_SILK_PLUS("GFA06", "Bambu PLA Silk+", FilamentType.PLA),
to Filament.java and rebuilt the project.