TFyre/bambu-farm #226: FQ: Add file storage

RyanEwen, 2025-02-08:
I think the main thing missing from this project is the ability for it to host gcode files rather than having to upload them each time one wants to print something to multiple machines.

I currently use SimplyPrint because it works well for this. I upload all of my commonly printed projects and send them to whichever printers I need at any given time, without having to reslice or reupload the files. It's great, and I think it's the main thing missing from bambu-farm :)

Aghost917, 2025-04-11:
This FQ was added in Version 1.6.0. Now it is no longer necessary to keep uploading and printing the same file over and over. Just have all of your commonly printed projects on the SD card.

RyanEwen, 2025-04-11:
Hmm but that then requires managing files on many SD cards (6 in my case) and that's a huge pain.

Aghost917, 2025-04-11:
> I upload all of my commonly printed projects and send them to whichever printers I need at any given time

Isn't that what batch print does? Because I already have my files on my server. 

> without having to reslice or reupload the files.

The “skip upload if file exists and is the same size” feature stops you from reuploading the files that you are printing. And if you resliced the file, you would need to reupload it to the printer to print anyways.

The part I’m still a bit confused about is how to host G-code files in the cloud in a way where they never need to be sent to the printer at all. I’m not quite sure how the cloud-hosted side of things works in that context.

RyanEwen, 2025-04-11:
Keeping in mind that I haven't seen any updates to bambu-farm since my OP...

My issue with how it worked when I tried it last, is that I had to be on the machine where the gcode files resided. That unfortunately isn't the Android tablet near the machines, or the laptop in the office next to them, or my laptop at home in the living room, etc. So while it is possible to pick files from the computer accessing bambu-farm, that is not very useful from other computers.

> The part I’m still a bit confused about is how to host G-code files in the cloud in a way where they never need to be sent to the printer at all. I’m not quite sure how the cloud-hosted side of things works in that context.

They would just get sent to the printer again for each print, this is how all the other options I've used work, such as SimplyPrint. The flaw is this method lies more on Bambu's shoulders than anyone else's IMO.

RyanEwen, 2025-04-11:
Maybe to avoid sending files that are already cached on the SD card, a simple hash check could be done to compare the hosted file to the cached file, and skip sending the file if they match.

Aghost917, 2025-04-11:
> Keeping in mind that I haven't seen any updates to bambu-farm since my OP...

Try out 1.6.0 it just got added yesterday.
I would suggest a server that has the files so your tablet, your computer, and your laptop can see it.

RyanEwen, 2025-04-12:
I had a play around with the newest and it appears to be the same workflow to me:

[image]

To make sure I'm not missing something I did choose a file from my main PC with all the sliced files on it, and when click Upload, choose a file, and press Print, it sends it to each printer. It didn't actually start the print, though, which was odd. Most importantly (within the context of this feature request) the uploaded file is not available to be chosen again from this screen (it's on each printer individually but not in bambu-farm to be bulk-printed)

What I am hoping can be added to bambu-farm is a simple file manage that shows gcode hosted on the same machine as bambu-farm, so that those files can be batch printed. For example, here's how SimplyPrint does it:

[image]

[image]

In this case the files are uploaded into SimplyPrint and stored in the cloud. They're available on any client and SimplyPrint sends them to each printer on every job.

Aghost917, 2025-04-12:
[image]

I think you are just wanting a cloud service to host your files. 

To host a file server from your computer, you can use software like Nextcloud, TrueNAS, or **built-in Windows features** to share files over a network. These options allow you to create a local file server accessible to other devices on the network. You can also use software like Nextcloud, TrueNAS, or Hyper-V role to run a Linux-based file server. 

> In this case the files are uploaded into SimplyPrint just once and stored in the cloud for reuse over and over. They're available on any client and SimplyPrint sends them to each printer on every job.

I would never have my files in the cloud so they can be stolen (I have a lot of people trying to steal my STL files). LAN only mode plus block from router from internet.

RyanEwen, 2025-04-12:
> I think you are just wanting a cloud service to host your files.

I'm wanting bambu-farm to be the file server, like most of the other farm software are.

> To host a file server from your computer, you can use software like Nextcloud, TrueNAS, or built-in Windows features ...

Users can absolutely use an existing cloud providers or setup their own file servers, but that is extra setup both to get the files there and to get access on every potential device that accesses bambu-farm. 

What would be better is if bambu-farm itself had access to a local directory on the server such as `/home/user/bambu-farm/gcode-files` and could display those files in the bambu-farm UI for printing. Similar to how it displays files on the SD cards, but from a local directory on the server. A bonus would be if it also allowed managing those files from the UI as well (upload, rename, delete, etc).

This would be much less clunky than the existing upload / browse input box which requires finding files from other third-party file systems and apps, which will differ per client device, and would require setup on every client device, which entirely defeats the purpose of a web-based system.

Hope this makes more sense.

I am a software developer myself by the way, however I haven't worked with Java in over a decade, otherwise I would consider doing it myself.

Aghost917, 2025-04-12:
> I'm wanting bambu-farm to be the file server, like most of the other farm software are.
>
>
> Users can absolutely use an existing cloud providers or setup their own file servers, but that is extra [...]

Ahhhhhhhh, I see what you want now. It does seem like a lot to add, though.