TFyre/bambu-farm #236: Feature Request: upload files to multiple printers

wouterokkerse, 2025-02-19:
Currently there is a option to upload a file to a printer but only one printer at a time and an option for upload and print where multiple printers can be selected. It would be nice if there is a feature to select the file to upload select the printers, but only upload without printing.

TFyre, 2025-04-10:
Please elaborate on the usecase? What is the purpose of uploading the file and not printing it?

Aghost917, 2025-04-11:
> Please elaborate on the usecase? What is the purpose of uploading the file and not printing it?

Use Case:
The goal of uploading files to printers while they’re already printing is to eliminate idle time and maximize efficiency in a print farm. Right now, having to wait for a printer to go idle just to upload files introduces unnecessary delays.

In my setup, I run my printers in 12-hour and 10-hour segments, and each printer uses over 25 different files and is always adding more. By uploading the next set of files while the current print is still running, I can have everything ready to go as soon as the current job finishes—no waiting around.

The only alternative right now is to physically remove every SD card, load the files onto each one manually, and then reinsert them back into the printers. That’s not only time-consuming and tedious but also introduces more risk for wear and tear on the cards and card slots.

On top of that, I like to keep everything organized on the SD card, with all related files grouped together. Since Bambu printers sort files by last modified time, I upload them in a specific order so that the 10- and 12-hour versions of the same model sit right next to each other on the interface. This makes it faster to locate the right file when starting the next job manually or via the new “skip upload if file exists and is the same size” feature (which works amazingly, FYI).

Being able to bulk upload without triggering a print—even while printers are busy—would be a huge improvement. It reduces downtime, avoids unnecessary SD card juggling, and keeps the workflow smooth and organized.

That would be _my_ use case—it might not be useful for everyone else, I don’t know. But it’d make a big difference in my setup.

wouterokkerse, 2025-05-20:
Here the use case is that we have a print farm where a lot of the same files get printed. Once in a while the file needs to be updated because of an update in the design.  Uploading per printer over ftp takes time. Taking all the SD cards out putting all those new files on and putting all those sd cards back also takes a lot of time. Having a button to just upload to all the printers makes it a lot easier.