TFyre/bambu-farm #242: Connection Refused Errors

mountainbiker54, 2025-03-12:
I'm getting the below errors, everything seems to be working with the exception of the live view (I don't have docker setup). The web page seems to be working just fine.  Are these issues that I should be worrying about/fixing? I'm wondering if this is because I have the X1E vs the X1C.

Both printers are on firmware version: 01.00.00.11

2025-03-12 14:08:30,008 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] ERROR [com.tfy.bam.pri.BambuPrinterStream] (vert.x-eventloop-thread-0) X1E-Left: clientFailure: io.netty.channel.AbstractChannel$AnnotatedConnectException - Connection refused: /x.x.x.143:6000
2025-03-12 14:08:30,009 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] ERROR [com.tfy.bam.pri.BambuPrinterStream] (vert.x-eventloop-thread-0) X1E-Right: clientFailure: io.netty.channel.AbstractChannel$AnnotatedConnectException - Connection refused: /x.x.x.119:6000

env file:
` bambu.cloud.enabled=true
bambu.cloud.username=xx_USERNAME_xx
bambu.cloud.token=xx_TOKEN_NUMBER_xx

quarkus.http.host=0.0.0.0
quarkus.http.port=8080
bambu.dark-mode=true
bambu.printers.myprinter1.name=X1E-Right
bambu.printers.myprinter1.device-id=xx-SN-RIGHT-xx
bambu.printers.myprinter1.access-code=xx_code_xx
bambu.printers.myprinter1.ip=x.x.x.119
bambu.printers.myprinter1.use-ams=true
bambu.printers.myprinter1.timelapse=true
bambu.printers.myprinter1.bed-levelling=true
bambu.printers.myprinter1.flow-calibration=true
bambu.printers.myprinter1.vibration-calibration=true

bambu.printers.myprinter2.name=X1E-Left
bambu.printers.myprinter2.device-id=xx-SN-LEFT-xx
bambu.printers.myprinter2.access-code=xx_code_xx
bambu.printers.myprinter2.ip=x.x.x.143
bambu.printers.myprinter2.use-ams=true
bambu.printers.myprinter2.timelapse=true
bambu.printers.myprinter2.bed-levelling=true
bambu.printers.myprinter2.flow-calibration=true
bambu.printers.myprinter2.vibration-calibration=true

bambu.auto-login=true
bambu.users.admin.password=admin
bambu.users.admin.role=admin
bambu.users.vzuser.password=1
bambu.users.vzuser.role=admin

bambu.use-bouncy-castle=true

quarkus.log.file.enable=true
quarkus.log.file.path=application.log
`

Application log: 
` 2025-03-12 14:07:17,699 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [io.und.servlet] (main) Initializing AtmosphereFramework
2025-03-12 14:07:17,743 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [com.vaa.flo.ser.sta.ServletDeployer] (main) Skipping automatic servlet registration because there is already a Vaadin servlet with the name com.vaadin.quarkus.QuarkusVaadinServlet
2025-03-12 14:07:17,751 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed AtmosphereHandler com.vaadin.flow.server.communication.PushAtmosphereHandler mapped to context-path: /*
2025-03-12 14:07:17,751 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed the following AtmosphereInterceptor mapped to AtmosphereHandler com.vaadin.flow.server.communication.PushAtmosphereHandler
2025-03-12 14:07:17,784 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) Atmosphere is using org.atmosphere.util.VoidAnnotationProcessor for processing annotation
2025-03-12 14:07:17,793 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.uti.ForkJoinPool] (main) Using ForkJoinPool  java.util.concurrent.ForkJoinPool. Set the org.atmosphere.cpr.broadcaster.maxAsyncWriteThreads to -1 to fully use its power.
2025-03-12 14:07:17,800 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) Installed WebSocketProtocol org.atmosphere.websocket.protocol.SimpleHttpProtocol 
2025-03-12 14:07:17,806 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.con.JSR356AsyncSupport] (main) JSR 356 Mapping path /VAADIN/push
2025-03-12 14:07:17,810 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [io.und.websockets] (main) UT026005: Adding programmatic server endpoint class org.atmosphere.container.JSR356Endpoint for path /VAADIN/push
2025-03-12 14:07:17,813 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [io.und.websockets] (main) UT026005: Adding programmatic server endpoint class org.atmosphere.container.JSR356Endpoint for path /VAADIN/push/
2025-03-12 14:07:17,813 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) Installing Default AtmosphereInterceptors
2025-03-12 14:07:17,813 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.CorsInterceptor : CORS Interceptor Support
2025-03-12 14:07:17,813 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.CacheHeadersInterceptor : Default Response's Headers Interceptor
2025-03-12 14:07:17,814 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.PaddingAtmosphereInterceptor : Browser Padding Interceptor Support
2025-03-12 14:07:17,814 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.AndroidAtmosphereInterceptor : Android Interceptor Support
2025-03-12 14:07:17,816 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.HeartbeatInterceptor : Heartbeat Interceptor Support
2025-03-12 14:07:17,816 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.SSEAtmosphereInterceptor : SSE Interceptor Support
2025-03-12 14:07:17,816 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.JSONPAtmosphereInterceptor : JSONP Interceptor Support
2025-03-12 14:07:17,817 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.JavaScriptProtocol : Atmosphere JavaScript Protocol
2025-03-12 14:07:17,817 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.WebSocketMessageSuspendInterceptor : org.atmosphere.interceptor.WebSocketMessageSuspendInterceptor
2025-03-12 14:07:17,817 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.OnDisconnectInterceptor : Browser disconnection detection
2025-03-12 14:07:17,817 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.cpr.AtmosphereFramework] (main) 	org.atmosphere.interceptor.IdleResourceInterceptor : org.atmosphere.interceptor.IdleResourceInterceptor
[... 61 lines cut ...]
2025-03-12 14:08:20,001 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] ERROR [com.tfy.bam.pri.BambuPrinterStream] (executor-thread-5) X1E-Right: No image received since 2025-03-12T14:07:21.131961847Z
2025-03-12 14:08:20,000 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] ERROR [com.tfy.bam.pri.BambuPrinterStream] (executor-thread-1) X1E-Left: No image received since 2025-03-12T14:07:22.448627665Z
2025-03-12 14:08:30,008 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] ERROR [com.tfy.bam.pri.BambuPrinterStream] (vert.x-eventloop-thread-0) X1E-Left: clientFailure: io.netty.channel.AbstractChannel$AnnotatedConnectException - Connection refused: /x.x.x.143:6000
2025-03-12 14:08:30,009 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] ERROR [com.tfy.bam.pri.BambuPrinterStream] (vert.x-eventloop-thread-0) X1E-Right: clientFailure: io.netty.channel.AbstractChannel$AnnotatedConnectException - Connection refused: /x.x.x.119:6000
2025-03-12 14:08:35,413 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [org.atm.uti.ForkJoinPool] (Shutdown thread) Using ForkJoinPool  java.util.concurrent.ForkJoinPool. Set the org.atmosphere.cpr.broadcaster.maxAsyncWriteThreads to -1 to fully use its power.
[... cut ...]
2025-03-12 14:08:35,650 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [com.tfy.bam.pri.BambuPrintersImpl] (Shutdown thread) X1E-Left: stopping
2025-03-12 14:08:35,650 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [com.tfy.bam.pri.BambuPrinterStream] (Shutdown thread) X1E-Left: stopping
2025-03-12 14:08:35,650 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [com.tfy.bam.pri.BambuPrintersImpl] (Shutdown thread) X1E-Left: stopped
2025-03-12 14:08:35,657 gen-mon01 /usr/lib/jvm/java-21-openjdk-amd64/bin/java[248796] INFO  [io.quarkus] (Shutdown thread) bambu-web stopped in 0.264s
`

[image]

TFyre, 2025-03-12:
https://github.com/TFyre/bambu-farm?tab=readme-ov-file#live-view

If im not mistaken X1C & X1E has the same interface for camera, so you should follow the docker setup if you want to have video

mountainbiker54, 2025-03-12:
Unfortunately that guide is lacking for someone who's never used docker before and only has a cursory understanding of linux in general. I feel like the draw the owl meme :(

[image]