TFyre/bambu-farm #245: Possible MQTT Problem

sswan940505, 2025-03-21:
Hi, first of all, thank you for this great open source project!

I successfully used two farm with 3 printers and 2 printers each.

I use docker for bambu farm, mediamtx, nginx, as you gave.

However, recently I add one more printer.

It's IP, access code, device name, etc. are all multiple checked.

And all X1E's firmware is 01.01.02.00.

However, 4'th X1E's state is always offline, and i cannot see the print progress, filaments, and cannot control.
Weird thing is, 4th X1E's camera is shown well, files can be uploaded to sd card.

What I have to check or do something to 4th X1E?

Thanks.

Aghost917, 2025-04-01:
I have the same problem with a P1S. Out of 32 printers, 1 is always offline by the time the printers finish printing. Only fix is restarting  bambu-web-1.5.0-runner.jar

TFyre, 2025-04-10:
Closing stale issue, please re-open if required

closed 2025-04-10 by TFyre